balena-os/meta-balena #672: The config.json is read multiple times whenever resin-vars is called

Page-, 2017-05-16:
Currently we read the config.json file for every variable in https://github.com/resin-os/meta-resin/blob/e4fcc8010beccdeaac468018511848c63809f6fa/meta-resin-common/recipes-support/resin-vars/resin-vars/resin-vars#L49-L58 - it would be much nicer if we only had one read into the file, and only spawned a jq process once

willnewton, 2017-05-31:
@Page- do you have an approach in mind for how to do this? The ways I can think of seem to be a bit messy and error prone, e.g. splitting on newlines in bash

Page-, 2017-06-15:
I don't have a specific approach in mind, but there are some suggestions in https://github.com/stedolan/jq/wiki/Cookbook#extract-data-and-set-shell-variables

willnewton, 2017-07-18:
Closed by https://github.com/resin-os/meta-resin/pull/759

closed 2017-07-18 by willnewton