balena-os/meta-balena #691: dnsmasq bind multible interface

pvizeli, 2017-06-08:
I see that dnsmasq bind do not bind only the `127.0.0.2`. That give a trouble with our dnsmasq hassio addon.

https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-connectivity/dnsmasq/files/dnsmasq.conf.systemd

I think that should be ```--listen-address``` instead `-a`?

agherzan, 2017-06-09:
Hi @pvizeli . `--listen-address` should be the log argument format of `-a` and both should have the same effect. How did you test that dnsmasq binds to something else?

floion, 2017-06-09:
You can look up the dnsmasq log and check there where it binds to, @pvizeli 
Where else do you suspect it is binding to?

pvizeli, 2017-06-18:
I'm sorry for late answer. This week was a bit hard on work.

[image]

So you see it listen on 0.0.0.0

agherzan, 2017-06-19:
Hi @pvizeli ,

Could you please provide us the output of `netstat -lntu` ? Dnsmasq should only bind to address 127.0.0.2 and I was able to check this on a RaspberryPi (resinOS 2.0.6). Mind the `-n` flag which will give you the actual IP. 

Regards,
Andrei

pvizeli, 2017-06-22:
You are right, it was my trouble on setup. Sorry for my noise and thanks for support.

closed 2017-06-22 by pvizeli