balena-os/meta-balena #712: Splashscreen in small screens

sGerli, 2017-06-18:
The splash screen gets cut in small screens, in mine I can only see the edge of the logo in the top left of the screen.

floion, 2017-06-19:
Hi, can you give more details on hardware and software you are experiencing this with?
What resinOS version? What board and what screen?

sGerli, 2017-06-19:
Sure, I'm using a Raspberry Pi 2 with a TFT 320x240px and Resin OS 2.0.6+rev3

floion, 2017-06-19:
Thanks. If you could also attach a photo here for reference, would be great.

sGerli, 2017-06-19:
Here's the photo
[image]

shaunmulligan, 2017-06-19:
I have seen this issue before, I think its because the device can't automatically detect the screens resolution, so you have to define the screen settings in the config.txt. There some good steps for doing this here: https://github.com/resin-io-playground/resin-wpe#displaying-on-a-pitft-28

sGerli, 2017-06-19:
I think that's a different problem, the screen works perfectly after that, it just looks cropped in the splashscreen

agherzan, 2017-07-18:
Hi @sGerli , did you manage to figure this out? What is the exact TFT model you are using?

sGerli, 2017-07-18:
No, my screen is a ILI9341

nmaas87, 2017-08-23:
I have the same problem as @sGerli with the same screen.
On bootup, the resin Logo is chopped.
I thought maybe setting the Framebuffer Size would help, but I did not.
Also hdmi_cvt does only change the resolution for the primary fb / HDMI out and setting this does not change anything for the secondary framebuffer / TFT Display.

ZubairLK, 2020-01-27:
resin-logo.png is 840x840 pixels by default.
I suspect this problem would go away if the logo file is replaced with a smaller one.
I would recommend doing that on a case by case basis.

Closing issue to cleanup. Can reopen if still relevant/needed

closed 2020-01-27 by ZubairLK