balena-os/meta-balena #721: Invalid MBR boot signature. Expected 0xAA55, but saw 0x0

brasilikum, 2017-06-28:
When I run

```sh
sudo resin local configure resin.img.zip
```

i get 

```sh
Invalid MBR boot signature. Expected 0xAA55, but saw 0x0

If you need help, don't hesitate in contacting us at:

  GitHub: https://github.com/resin-io/resin-cli/issues/new
  Forums: https://forums.resin.io
```

➜ resin version
6.0.0

➜ image version
2.0.6 rev1

➜ device
raspberry pi 2

Would be nice if someone could help me debug this

zvin, 2017-06-28:
I'm not 100% sure but I think `resin local configure` works on raw images.
You'll need to unzip `resin.img.zip` first.

agherzan, 2017-07-18:
@brasilikum Can you confirm that after unpacking the archive you are able to use the tool?

brasilikum, 2017-07-24:
I can!

closed 2017-07-24 by brasilikum