balena-os/meta-balena #727: Module v4l2loopback not found

zupan, 2017-07-02:
Hello,
I'm trying to create v4l2 loopback devices on NUC deployed through resin.io and I'm not able to.
After installing all necesary packages:
```
sudo apt-get install v4l2loopback-dkms v4l-utils ffmpeg
sudo modprobe v4l2loopback devices=2
v4l2-ctl --list-devices
```
device outputs an error
```
modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/4.8.17-yocto-standard
```
I'm using ubuntu:16.04 docker image.
Posted procedure works on normal ubuntu instalation, but I'm wondering if someone was able to make it work through resin? Or if there is a known solution for setting module?

agherzan, 2017-07-11:
ResinOS doesn't ship this kernel module. As far as I can see this is not in mainline and you will need to compile it using the provided resinOS kernel headers. See https://github.com/resin-io-playground/kernel-module-build

agherzan, 2017-07-18:
I will close this issue for now. Feel free to comment if we can assist you further.

closed 2017-07-18 by agherzan