balena-os/meta-balena #736: resin-sample wifi connection poses a possible security risk

kipe, 2017-07-06:
At the current state, the device connects to the example wifi profile `resin-sample` (if for example the attacker mimics the example network). This poses a security risk, as the device connects automatically to a network which the user hasn't defined.

Possible fix could be to add `autoconnect = false` in `[connection]` block. However this also requires comments, so that the user is sure set the value to `true`.

agherzan, 2017-07-06:
Hi @kipe . Can describe a little bit the attack vector? As well, is this related resinOS managed or unmanaged?

kipe, 2017-07-06:
Hey @agherzan .

The attack vector is fairly simple, providing these requirements are met:

- Device either disconnected completely or connected via ethernet
- [resin-sample](https://github.com/resin-os/meta-resin/blob/d509830d673e1c819bbd289facce6f50da69ad72/meta-resin-common/recipes-connectivity/networkmanager/files/resin-sample) file hasn't been modified
- The attacker configures an access point to mimic the configuration in `resin-sample`. This means matching SSID and password of `resin-sample`.

If I for example setup my laptop to mimic the wifi access point defined `resin-sample`, the device (RPi3 in my case) automatically connects to the access point. This naturally exposes also all the listening services defined both by the users container and the host OS to the laptop. This for example could allow access to sensitive data or control of the device.

As far as I understand, this relates to both managed and unmanaged versions. Tested with a RPi3 connected to resin.io production environment, running `Resin OS 2.0.3+rev1 (prod)` and supervisor `4.2.2`.

agherzan, 2017-07-06:
That clarifies it. Thanks @kipe . We already have a bug submited in hq repository as Greg referenced above. I will close this issue as duplicate and let's follow up the discussion there. Thanks once again.

agherzan, 2017-07-17:
I realized that the repository I'm referring to above is not public and we closed this issue without any further information. I will reopen it and we will keep it updated as it will progress.

willnewton, 2017-08-09:
This is fixed in the 2.2.0 release.

closed 2017-08-09 by willnewton