balena-os/meta-balena #738: .../meta-resin/meta-resin-common/classes/resin-go.bbclass:1: Could not inherit file classes/go.bbclass

gmacario, 2017-07-08:
Hello,

I have been trying to add meta-resin to the GENIVI Development Platform, a Yocto distro currently based on morty.

I have added  `meta-resin/meta-resin-common` and `meta-resin/meta-resin-morty`, but when I try to bitbake my custom image I receive the following error:

```
+ bitbake rokers-image-base
Parsing recipes...ERROR: ParseError at /var/jenkins_home/workspace/dev-platform_wip-add-docker-RN5PWYXJBMW7YIYKXAV7CAJHUYXGQ74ZGGLPQYJRPF6G5G47BFJA/gdp-src-build/../meta-resin/meta-resin-common/classes/resin-go.bbclass:1: Could not inherit file classes/go.bbclass

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

Any suggestions about what I did wrong?

gmacario, 2017-07-08:
The error seems caused by `resin.go.bbclass` which cannot inherit file `classes/go.bbclass`.

My distro already includes the `poky` layer (branch morty).
Unfortunately `poky/meta/classes/go.bbclass` only exists on https://git.yoctoproject.org/cgit.cgi/poky/tree/meta/classes/go.bbclass
but is not present on the `morty` branch which I have been using so far.
Notice that I cannot update the branch of poky right now.

willnewton, 2017-07-10:
At the moment the Go support recipes are taken from the oe-meta-go repository e.g. see https://github.com/resin-os/resin-raspberrypi/tree/master/layers

floion, 2017-07-10:
Also, please checkout the oe-meta-go at commit a15be494a64c36fd7dc4d1808c9d13d0fbee56bf

agherzan, 2017-07-10:
Hi @gmacario ,

This is because there is a dependency on oe-meta-go layer which is not documented properly. I opened a PR to add this information in the README.

Regards,
Andrei

gmacario, 2017-07-10:
Thanks @agherzan, I will add oe-meta-go per your suggestion and will update the ticket as soon as the build completes successfully

gmacario, 2017-07-10:
I confirm that adding submodule `oe-meta-go` (commit a15be494a64c36fd7dc4d1808c9d13d0fbee56bf) as suggested fixed the issue for me.

closed 2017-07-10 by gmacario