balena-os/meta-balena #773: Include deviceType in the default config.json?

pcarranzav, 2017-07-27:
The supervisor, even in offline mode, needs to know the deviceType and currently only knows about it using config.json.
It should be included in https://github.com/resin-os/meta-resin/blob/a98e4fb3a0298370cb25f1c60f2204a6bfce986f/meta-resin-common/classes/image-resin.bbclass#L38 so that the supervisor works properly when in offline mode ( e.g. loading preloaded apps) without the need to modify config.json. Is this feasible?

agherzan, 2017-07-28:
@pcarranzav Sadly, as far as I remember, the config.json generated by yocto is completely overwritten by image maker. The latter doesn't add to that file but replaces it with a new file. @shaunmulligan can you confirm? If that is the case, this task should be in the image maker.

shaunmulligan, 2017-07-28:
But @agherzan isn't the offline supervisor mode for resinOS unmanaged devices? Meaning there is no image maker interaction at all. The image maker only happens on resin.io images

agherzan, 2017-07-28:
I didn't know this issue was unmanaged specific, was it? I though by `offline mode` Pablo meant disconnected from the network not unmanaged. If that is the case, we can definitely do it.

pcarranzav, 2017-07-28:
Yeah, the supervisor works in "offline mode" when on an unmanaged device. So this is about the config.json when there's no image maker, sorry for not being clear

agherzan, 2017-07-28:
Sounds good. Thanks for clarifying.

willnewton, 2017-07-31:
Opened a PR here: https://github.com/resin-os/meta-resin/pull/780

willnewton, 2017-08-21:
This should be done in 2.3.0

closed 2017-08-21 by willnewton