balena-os/meta-balena #774: Persistent logs don't seem to be persisting the full 8MB of logs before rotation

Page-, 2017-07-27:


agherzan, 2017-07-28:
@willnewton could you please take a look on this?

@Page- the front links are not available to external users. Could you summarise its content in the descriptions of this issue? Thanks.

willnewton, 2017-07-28:
For reference it appears the device is running resinOS 2.0.0+rev3.

willnewton, 2017-07-28:
I can't see any obvious problems with how things are setup now. It would be useful to get a concise problem statement e.g. is it the start of the logs that are missing, the end or gaps in the middle.

floion, 2017-08-21:
I have another report that is basically saying logs are not persisted over reboots

agherzan, 2017-08-21:
@floion can you please link the report you are referring to?

ZubairLK, 2019-01-22:
closing to clean up

closed 2019-01-22 by ZubairLK