balena-os/meta-balena #800: Custom openvpn client?

xanview, 2017-08-14:
What is the recommended way to run a custom OpenVPN client, building a new image from scratch with Yoctu/bitbake, or is there some easier way?

agherzan, 2017-08-21:
Hello @xanview . At this point we don't support custom openvpn client configuration modular or as a feature in resinOS. This means that at this point, indeed, the only way forward would be a custom yocto build which would include the correct configuration and/or other customisation.

xanview, 2017-08-22:
Ah, thank you for your reply. Do you have a script or some documentation somewhere how to build a ResinOS yocto build?

agherzan, 2017-08-22:
What board are you interested in? Are you familiar with yocto?

xanview, 2017-08-22:
Raspberry Pi3, NUC and Asus Tinkerboard, I saw there's a meta image for Tinkerboard. I'm not familiar with Yocto, going to do a bit of reading about it.

agherzan, 2017-08-23:
When you say a custom openvpn client, do you mean just a custom openvpn client configuration? Could you detail the use case?

xanview, 2017-08-24:
Yes, sorry, just a custom config, for example:

```
client
dev tap
proto tcp
remote custom-domain 1194 udp
remote custom-domain 443 tcp-client
resolv-retry infinite
nobind
persist-key
persist-tun
comp-lzo
ca /wherever/custom-domain_ca.crt
cert /wherever/custom-domain_client.crt
key /wherever/custom-domain_client.key
ns-cert-type server
verb 3
mute 20
```

Several reasons:
- UDP has a smaller overhead and more efficient to use on cellular connections
- Some customers want to be in full control of all communication, e.g. financial institutions
- To connect to a custom cloud server, e.g. what we do here with surveillance cameras as failover communication/reverse proxy: https://monitor.timeline.is

agherzan, 2017-08-29:
Hello again,

We had an internal discussion about this and we concluded that it is a feature we would like to have in resinOS. By "this" I mean a way of having custom configuration for openvpn in resinOS. The problem is that currently we don't have the time to look into it and it doesn't fit into our high priority tasks. What we can offer though is support if you are willing to invest some time into developing this as a core feature of resinOS. The implementation would probably be something similar to what we have for proxy feature https://github.com/resin-os/meta-resin/pull/618 .

Regards,
Andrei

ZubairLK, 2020-01-27:
Closing old/stale issues. Might still be valid. Can reopen if needed.

closed 2020-01-27 by ZubairLK