balena-os/meta-balena #801: Docker dies: Error initializing network controller: Error creating default \"bridge\" network: cannot create  network docker0

shaunmulligan, 2017-08-14:
**Device type:** TS-4900
**OS version:** Resin OS 2.0.6+rev1 (dev)
```
Aug 14 18:00:50 730efa3 systemd[1]: [[0;1;39mdocker.service: Unit entered failed state.[[0m
Aug 14 18:00:50 730efa3 systemd[1]: [[0;1;39mdocker.service: Failed with result 'exit-code'.[[0m
Aug 14 18:00:50 730efa3 systemd[1]: docker.service: Service hold-off time over, scheduling restart.
Aug 14 18:00:50 730efa3 systemd[1]: Stopped Docker Application Container Engine.
Aug 14 18:00:50 730efa3 systemd[1]: Starting Docker Application Container Engine...
Aug 14 18:00:51 730efa3 docker[903]: time="2017-08-14T18:00:51.121029667Z" level=warning msg="/!\\ DON'T BIND ON ANY IP ADDRESS WITHOUT setting -tlsverify IF YOU DON'T KNOW WHAT YOU'RE DOING /!\\"
Aug 14 18:00:51 730efa3 docker[903]: time="2017-08-14T18:00:51.185783334Z" level=info msg="Graph migration to content-addressability took 0.00 seconds"
Aug 14 18:00:51 730efa3 docker[903]: time="2017-08-14T18:00:51.207820001Z" level=info msg="Firewalld running: false"
Aug 14 18:00:51 730efa3 docker[903]: time="2017-08-14T18:00:51.730166001Z" level=info msg="Default bridge (docker0) is assigned with an IP address 172.17.0.0/16. Daemon option --bip can be used to set a preferre
d IP address"
Aug 14 18:00:51 730efa3 docker[903]: time="2017-08-14T18:00:51.931294334Z" level=fatal msg="Error starting daemon: Error initializing network controller: Error creating default \"bridge\" network: cannot create 
network docker0 (d4504fc3886a40e9fc1b3b6989b7dd9f90daa02ffe50dc5a69e547990a7ce287): conflicts with network febeb99d1d60938eff1bde7dde65b61e1d872cc8cfe4eba9006a3825790df124 (docker0): networks have same bridge na
me"
Aug 14 18:00:51 730efa3 systemd[1]: [[0;1;39mdocker.service: Main process exited, code=exited, status=1/FAILURE[[0m
Aug 14 18:00:51 730efa3 systemd[1]: [[0;1;31mFailed to start Docker Application Container Engine.[[0m
Aug 14 18:00:52 730efa3 systemd[1]: [[0;1;39mdocker.service: Unit entered failed state.[[0m
Aug 14 18:00:52 730efa3 systemd[1]: [[0;1;39mdocker.service: Failed with result 'exit-code'.[[0m
Aug 14 18:00:52 730efa3 systemd[1]: docker.service: Service hold-off time over, scheduling restart.
```

shaunmulligan, 2017-08-14:
seems related to this: https://github.com/moby/moby/issues/17083 with the fix being: `sudo rm -rf /var/lib/docker/network/files` on the hostOS

shaunmulligan, 2017-08-14:
I got the device into this state by rebooting it from the dashboard, so I guess docker is not properly cleaning up its networks at shutdown.

agherzan, 2017-08-21:
The reboot device action reboots the host OS, right? This means that at clean reboot the device couldn't start docker?

shaunmulligan, 2017-08-21:
@agherzan yeah, basically dockers graceful shutdown is not cleaning up correctly. But this has been fixed in 17.03 as far as I can see.

willnewton, 2017-10-11:
In resinOS 2.0.7 we updated to Docker 17.03.1. Does that mean we would expect this issue to be fixed @shaunmulligan ?

shaunmulligan, 2017-10-11:
@willnewton yeah that is my understanding of it currently.

willnewton, 2017-10-11:
@shaunmulligan is the way to reproduce to reboot from the dashboard a large number of times?

shaunmulligan, 2017-10-11:
@willnewton yeah as far as I understand, but it might only have happened on the TS4900, I haven't ever seen it on the RPI family

agherzan, 2018-02-07:
@shaunmulligan is this still reproducible? Was it fixed with balena?

imrehg, 2018-02-10:
@agherzan don't think we've seen this lately and no reports, thus pointing to this not being an issue anymore. Probably good to close now (and reopen if new info comes in)?

agherzan, 2018-02-11:
Sounds good.

closed 2018-02-11 by agherzan