balena-os/meta-balena #812: Network bonding does not work on resinOS for the raspberry pi

josephroberts, 2017-08-23:
Once a network bonding connection has been configued the wlan0 interface constantly changes state between up and down.

```
[  221.450505] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[  221.463964] cfg80211: World regulatory domain updated:
[  221.476026] cfg80211:  DFS Master region: unset
[  221.480549] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  221.504399] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  221.519556] cfg80211:   (2457000 KHz - 2482000 KHz @ 20000 KHz, 92000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  221.536193] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[  221.551359] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  221.567994] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  221.584514] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[  221.599638] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[  221.614565] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[  221.637127] merck: link status definitely down for interface wlan0, disabling it
[  222.247059] merck: link status definitely up for interface wlan0, 0 Mbps full duplex
```

the wlan0 interface also does not start correctly on boot
```
root@440eecc:~# journalctl -u NetworkManager | grep -e wlan0
Aug 15 13:50:59 440eecc NetworkManager[657]: <info>  [1502805059.6753] keyfile: new connection /etc/NetworkManager/system-connections/bond-slave-wlan0 (bfbf0ba8-29d0-4040-a345-1b1a247871ae,"bond-slave-wlan0")
Aug 15 13:51:00 440eecc NetworkManager[657]: <info>  [1502805060.5653] wifi-nl80211: (wlan0): using nl80211 for WiFi device control
Aug 15 13:51:00 440eecc NetworkManager[657]: <info>  [1502805060.5663] device (wlan0): driver supports Access Point (AP) mode
Aug 15 13:51:00 440eecc NetworkManager[657]: <info>  [1502805060.5703] manager: (wlan0): new 802.11 WiFi device (/org/freedesktop/NetworkManager/Devices/3)
Aug 15 13:51:00 440eecc NetworkManager[657]: <info>  [1502805060.5742] device (wlan0): state change: unmanaged -> unavailable (reason 'managed') [10 20 2]
Aug 15 13:51:01 440eecc NetworkManager[657]: <info>  [1502805061.1612] device (wlan0): set-hw-addr: set MAC address to BE:B7:E6:43:83:EE (scanning)
Aug 15 13:51:01 440eecc NetworkManager[657]: <info>  [1502805061.5008] device (wlan0): supplicant interface state: init -> starting
Aug 15 13:51:01 440eecc NetworkManager[657]: <info>  [1502805061.5870] sup-iface[0x53d818,wlan0]: supports 5 scan SSIDs
Aug 15 13:51:01 440eecc NetworkManager[657]: <info>  [1502805061.5897] device (wlan0): supplicant interface state: starting -> ready
Aug 15 13:51:01 440eecc NetworkManager[657]: <info>  [1502805061.5899] device (wlan0): state change: unavailable -> disconnected (reason 'supplicant-available') [20 30 42]
Aug 15 13:51:02 440eecc NetworkManager[657]: <info>  [1502805062.1209] device (wlan0): supplicant interface state: ready -> inactive
Aug 15 13:51:02 440eecc NetworkManager[657]: <info>  [1502805062.4348] policy: auto-activating connection 'bond-slave-wlan0'
Aug 15 13:51:02 440eecc NetworkManager[657]: <info>  [1502805062.4589] device (wlan0): Activation: starting connection 'bond-slave-wlan0' (bfbf0ba8-29d0-4040-a345-1b1a247871ae)
Aug 15 13:51:02 440eecc NetworkManager[657]: <info>  [1502805062.4604] device (wlan0): state change: disconnected -> prepare (reason 'none') [30 40 0]
Aug 15 13:51:02 440eecc NetworkManager[657]: <info>  [1502805062.9702] device (wlan0): set-hw-addr: reset MAC address to B8:27:EB:F6:6C:EC (preserve)
Aug 15 13:51:03 440eecc NetworkManager[657]: <info>  [1502805063.0063] device (wlan0): supplicant interface state: inactive -> disabled
Aug 15 13:51:03 440eecc NetworkManager[657]: <info>  [1502805063.0080] device (wlan0): supplicant interface state: disabled -> inactive
Aug 15 13:51:03 440eecc NetworkManager[657]: <info>  [1502805063.0124] device (wlan0): state change: prepare -> config (reason 'none') [40 50 0]
Aug 15 13:51:03 440eecc NetworkManager[657]: <info>  [1502805063.0135] device (wlan0): Activation: (wifi) access point 'bond-slave-wlan0' has security, but secrets are required.
Aug 15 13:51:03 440eecc NetworkManager[657]: <info>  [1502805063.0137] device (wlan0): state change: config -> need-auth (reason 'none') [50 60 0]
Aug 15 13:51:03 440eecc NetworkManager[657]: <info>  [1502805063.0367] device (wlan0): state change: need-auth -> prepare (reason 'none') [60 40 0]
Aug 15 13:51:03 440eecc NetworkManager[657]: <info>  [1502805063.0389] device (wlan0): state change: prepare -> config (reason 'none') [40 50 0]
Aug 15 13:51:03 440eecc NetworkManager[657]: <info>  [1502805063.0401] device (wlan0): Activation: (wifi) connection 'bond-slave-wlan0' has security, and secrets exist.  No new secrets needed.
Aug 15 13:51:03 440eecc NetworkManager[657]: <info>  [1502805063.3022] device (wlan0): supplicant interface state: inactive -> associating
Aug 15 13:51:03 440eecc NetworkManager[657]: <info>  [1502805063.4323] device (wlan0): supplicant interface state: associating -> 4-way handshake
```

To reproduce, run the following commands from the hostOS and look at the dmesg/journalctl logs
```
// Create the bond connection
nmcli con add type bond ifname merck bond.options "mode=balance-rr,miimon=100"

// Assign ethernet slave
nmcli con add type ethernet ifname eth0 master merck

// Assign wifi slave
nmcli con add type wifi ssid <SSID> ifname wlan0 master merck
nmcli con mod bond-slave-wlan0 wifi-sec.key-mgmt wpa-psk
nmcli con mod bond-slave-wlan0 wifi-sec.psk <PASSWORD>

// Activate
nmcli con up bond-slave-eth0
nmcli con up bond-slave-wlan0
```

I noticed that the brcmfmac firmware versions are different
resinOS `brcmfmac: Firmware version = wl0: Aug 29 2016 20:48:16 version 7.45.41.26 (r640327) FWID 01-4527cfab`
Raspbian `brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378`

This is fixed by copying the contents of `/lib/firmware` from a raspbian image into `/resin-rootA/lib/firmware` on a resinOS image. Specifically these files get overwritten
```
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/ar7010_1_1.fw'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/ar7010.fw'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/ar9271.fw'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/brcm/brcmfmac43143-sdio.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/brcm/brcmfmac43143.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/brcm/brcmfmac43430-sdio.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/brcm/brcmfmac43430-sdio.txt'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/htc_7010.fw'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/htc_9271.fw'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rt2561.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rt2561s.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rt2661.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rt2860.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rt2870.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rt3070.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rt3071.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rt3090.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rt3290.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rt73.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rtlwifi/rtl8188eufw.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rtlwifi/rtl8192cufw.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rtlwifi/rtl8192cufw_A.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rtlwifi/rtl8192cufw_B.bin'? y
cp: overwrite '/run/media/joe/resin-rootA/lib/firmware/rtlwifi/rtl8192cufw_TMSC.bin'? y
```

You can test the bond interface is working correctly with `watch -d -n1 cat /proc/net/bonding/merck`

agherzan, 2017-09-04:
@josephroberts I know you narrowed down the problem. Could you add a comment here about your findings?

josephroberts, 2017-09-04:
@agherzan pretty sure it's just a case of making sure updating the `brcmfmac` firmware to `brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378`

agherzan, 2017-09-14:
Will be fixed when https://github.com/resin-os/resin-raspberrypi/issues/121 gets fixed (firmware updated).

closed 2018-02-02 by agherzan