balena-os/meta-balena #816: Make sure fsck is ran on all rw partitions

agherzan, 2017-08-28:
We currently run fsck for data partition. Conf and boot need to be checked too as they are writable partitions too. 

Connects to https://github.com/resin-io/hq/issues/999

willnewton, 2017-08-31:
I believe systemd also by default fscks the root partition.

I have opened a PR for fscking the resin-state partition here:

https://github.com/resin-os/meta-resin/pull/824

The remaining interesting partition is resin-boot but unfortunately mtools, which we use as part of our build process, creates slightly corrupt filesystem images which the standard fsck.fat will then delete the system-connections directory.

I've opened a PR upstream to try and resolve this:

https://github.com/dosfstools/dosfstools/pull/61

willnewton, 2017-10-27:
The fsck change has been merged so we can probably start to fsck the FAT partition now.

commit da1368b8 (2018-02-01): resin-image-initramfs: Reflect initramfs module rename

Fixes #816

Change-type: patch
Changelog-entry: Add fs check for boot partition in initramfs
commit f945798e (2018-02-05): resin-image-initramfs: Reflect initramfs module rename

Fixes #816

Change-type: patch
Changelog-entry: Add fs check for boot partition in initramfs
PR #968 (2018-02-05): FS check on boot partition
Fixes #816

---
### Contributor checklist

- [x] Changes have been tested
- [x] `Change-type` present on at least one commit
- [x] `Signed-off-by` is present
- [x] The PR complies with the [Open Embedded Commit Patch Message Guidelines](http://www.openembedded.org/wiki/Commit_Patch_Message_Guidelines)

### Reviewer Guidelines
- When submitting a review, please pick:
  - '*Approve*' if this change would be acceptable in the codebase (even if there are minor or cosmetic tweaks that could be improved).
  - '*Request Changes*' if this change would not be acceptable in ou [... cut]
commit f6810415 (2018-02-05): resin-image-initramfs: Reflect initramfs module rename

Fixes #816

Change-type: patch
Changelog-entry: Add fs check for boot partition in initramfs
closed 2018-02-05 by agherzan