balena-os/meta-balena #830: VPN restarted and resin-vpn interface recreated, happens about 30 seconds after boot

shaunmulligan, 2017-09-01:
In newer (2.2 and above) images I have noticed that everytime the device boots up, it show up on the dashboard, then drops offline 30 seconds later and is offline for ~10 - 15 seconds and then comes back online as usual.

This is a snap shot of the openvpn-resin service logs when that event happens:
```
Aug 31 16:50:33 e422fa7 openvpn[735]: VERIFY OK: depth=0, C=UK, ST=UK, L=London, O=Resin, OU=ResinVPN, CN=server, name=EasyRSA, emailAddress=hello@resin.io
Aug 31 16:50:34 e422fa7 openvpn[735]: Data Channel Encrypt: Cipher 'BF-CBC' initialized with 128 bit key
Aug 31 16:50:34 e422fa7 openvpn[735]: Data Channel Encrypt: Using 160 bit message hash 'SHA1' for HMAC authentication
Aug 31 16:50:34 e422fa7 openvpn[735]: Data Channel Decrypt: Cipher 'BF-CBC' initialized with 128 bit key
Aug 31 16:50:34 e422fa7 openvpn[735]: Data Channel Decrypt: Using 160 bit message hash 'SHA1' for HMAC authentication
Aug 31 16:50:34 e422fa7 openvpn[735]: Control Channel: TLSv1, cipher TLSv1/SSLv3 DHE-RSA-AES256-SHA, 2048 bit RSA
Aug 31 16:50:34 e422fa7 openvpn[735]: [server] Peer Connection Initiated with [AF_INET]52.5.48.142:443
Aug 31 16:50:36 e422fa7 openvpn[735]: SENT CONTROL [server]: 'PUSH_REQUEST' (status=1)
Aug 31 16:50:36 e422fa7 openvpn[735]: PUSH: Received control message: 'PUSH_REPLY,route 52.4.252.97,sndbuf 0,rcvbuf 0,ping 10,ping-restart 60,socket-flags TCP_NODELAY,ifconfig 10.2.15.124 52.4.252.97'
Aug 31 16:50:36 e422fa7 openvpn[735]: OPTIONS IMPORT: timers and/or timeouts modified
Aug 31 16:50:36 e422fa7 openvpn[735]: OPTIONS IMPORT: --sndbuf/--rcvbuf options modified
Aug 31 16:50:36 e422fa7 openvpn[735]: Socket Buffers: R=[341760->341760] S=[44800->44800]
Aug 31 16:50:36 e422fa7 openvpn[735]: OPTIONS IMPORT: --socket-flags option modified
Aug 31 16:50:36 e422fa7 openvpn[735]: Socket flags: TCP_NODELAY=1 succeeded
Aug 31 16:50:36 e422fa7 openvpn[735]: OPTIONS IMPORT: --ifconfig/up options modified
Aug 31 16:50:36 e422fa7 openvpn[735]: OPTIONS IMPORT: route options modified
Aug 31 16:50:36 e422fa7 openvpn[735]: ROUTE_GATEWAY 10.0.0.1/255.255.0.0 IFACE=eth0 HWADDR=b8:27:eb🇩🇪65:2d
Aug 31 16:50:36 e422fa7 openvpn[735]: TUN/TAP device resin-vpn opened
Aug 31 16:50:36 e422fa7 openvpn[735]: TUN/TAP TX queue length set to 100
Aug 31 16:50:36 e422fa7 openvpn[735]: do_ifconfig, tt->ipv6=0, tt->did_ifconfig_ipv6_setup=0
Aug 31 16:50:36 e422fa7 openvpn[735]: /sbin/ip link set dev resin-vpn up mtu 1500
Aug 31 16:50:36 e422fa7 openvpn[735]: /sbin/ip addr add dev resin-vpn local 10.2.15.124 peer 52.4.252.97
Aug 31 16:50:36 e422fa7 openvpn[735]: /etc/openvpn/upscript.sh resin-vpn 1500 1544 10.2.15.124 52.4.252.97 init
Aug 31 16:50:36 e422fa7 openvpn[735]: /sbin/ip route add 52.4.252.97/32 via 52.4.252.97
Aug 31 16:50:36 e422fa7 openvpn[735]: ERROR: Linux route add command failed: external program exited with error status: 2
Aug 31 16:50:36 e422fa7 openvpn[735]: GID set to openvpn
Aug 31 16:50:36 e422fa7 openvpn[735]: UID set to openvpn
Aug 31 16:50:36 e422fa7 openvpn[735]: Initialization Sequence Completed
Aug 31 16:52:38 e422fa7 openvpn[735]: [server] Inactivity timeout (--ping-restart), restarting
Aug 31 16:52:38 e422fa7 openvpn[735]: /etc/openvpn/downscript.sh resin-vpn 1500 1544 10.2.15.124 52.4.252.97 restart
Aug 31 16:52:38 e422fa7 openvpn[735]: WARNING: Failed running command (--up/--down): external program exited with error status: 1
Aug 31 16:52:38 e422fa7 openvpn[735]: Exiting due to fatal error
Aug 31 16:52:38 e422fa7 openvpn[735]: Closing TUN/TAP interface
Aug 31 16:52:38 e422fa7 openvpn[735]: /sbin/ip addr del dev resin-vpn local 10.2.15.124 peer 52.4.252.97
Aug 31 16:52:38 e422fa7 openvpn[735]: Linux ip addr del failed: external program exited with error status: 2
Aug 31 16:52:38 e422fa7 openvpn[735]: /etc/openvpn/downscript.sh resin-vpn 1500 1544 10.2.15.124 52.4.252.97 init
Aug 31 16:52:38 e422fa7 systemd[1]: openvpn-resin.service: Main process exited, code=exited, status=1/FAILURE
Aug 31 16:52:38 e422fa7 systemd[1]: openvpn-resin.service: Unit entered failed state.
Aug 31 16:52:38 e422fa7 systemd[1]: openvpn-resin.service: Failed with result 'exit-code'.
Aug 31 16:52:48 e422fa7 systemd[1]: openvpn-resin.service: Service hold-off time over, scheduling restart.
Aug 31 16:52:48 e422fa7 systemd[1]: Stopped OpenVPN Application On Resin.
Aug 31 16:52:48 e422fa7 systemd[1]: Starting OpenVPN Application On Resin...
```

agherzan, 2017-09-05:
Tried to replicate on 2.3.0 (Not reproducible). Going to try to latest one.

shaunmulligan, 2017-09-05:
@agherzan I see this consistently on an rpi3 2.3.0, did you not see the device go offline on the dashboard for a little time at boot?

agherzan, 2017-09-05:
I tested it on qemu. Both 2.3.0 and 2.4.2 versions have absolutely no VPN downtime at boot. Do you suspect something machine dependent?

shaunmulligan, 2017-09-05:
Yah, maybe it's some how rpi3 specific, but Craig, myself and @floion have seen this

agherzan, 2017-09-05:
OK. I'll try it tomorrow on RPI3 too.

agherzan, 2017-09-06:
@shaunmulligan I tried to reproduce this on current production images with version 2.3.0, RaspberryPi 3 device type and I could not reproduce not even once. Could this have been caused by a VPN server issue? I don't know how to proceed. Can you try to reproduce this on a fresh device? There is absolutely nothing I could see.

agherzan, 2017-09-06:
Ping @floion and @craig-mulligan because as far as I understood you were able to reproduce it too. Maybe there is more information I can get for any of you.

agherzan, 2017-09-11:
Ping @shaunmulligan @floion @craig-mulligan

agherzan, 2017-10-30:
I will close this issue for now. Feel free to reopen if you guys have any additional information.

closed 2017-10-30 by agherzan