balena-os/meta-balena #831: NetworkManager Blocks Until Timeout During Startup // Fixed with gnutls 3.5.9

gq0, 2017-09-04:
Greetings everyone,

The first attempt to launch the NetworkManager during startup failed with a timeout after 1.5 minutes which unfortunately delayed the readiness of our device. 

Recently we upgraded to **meta-resin 2.4.2** with the **NetworkManager** in version **1.8.2**. The issue was that the NetworkManager loads the **gnutls** library which does on initialisation a ``getrandom()`` call. This call blocks until the entropy level is high enough in the system to initialise the nonblocking random number pool; please see [0]. This is fixed in a newer version of gnutls [1] which isn't available in **morty** yet. Therefore, we upgraded manually to **gnutls 3.5.9** the default in **pyro**.

Maybe this is also something for your **meta-resin-morty** layer.

[0] https://github.com/systemd/systemd/issues/4271
[1] https://gitlab.com/gnutls/gnutls/merge_requests/111/

agherzan, 2017-09-04:
Thanks @gq0 for your report on this. Would you be able to push a PR which updates the gnutls package?

gq0, 2017-09-04:
Yes, please have a look #832 I just copied our stuff into your layer.

willnewton, 2017-10-18:
This PR has been merged.

closed 2017-10-18 by willnewton