balena-os/meta-balena #844: Harden ResinOS kernel

afitzek, 2017-09-13:
We should activate some hardening kernel configuration, here is a list of kernel hardening configuration options for the kernel: https://kernsec.org/wiki/index.php/Kernel_Self_Protection_Project/Recommended_Settings

I want to highlight: 
- CONFIG_CC_STACKPROTECTOR=y
- CONFIG_CC_STACKPROTECTOR_STRONG=y

FD: https://www.flowdock.com/app/rulemotion/r-security/threads/ZXIGjUS1p-jgJ7bSXcUSIb-CaIc

willnewton, 2017-09-13:
See #846

willnewton, 2017-09-19:
@afitzek the stack protector options have been enabled. I'm not sure if there are any other items in the list that you think would be good to enable?

It's not always easy to tell how restrictive an option might be for users or how it might affect performance.

agherzan, 2018-02-01:
Ping @afitzek . Given that stack protector options where enabled, can we close this issue?

agherzan, 2018-02-06:
Closing for now.

closed 2018-02-06 by agherzan