balena-os/meta-balena #858: Update dropbear to v2017.75

hellais, 2017-09-22:
It seems like the latest image of resin-os comes with an old version of dropbear.

The latest version of dropbear includes a fix for a critical security issue (remote code exec). See their changelog for more details: https://matt.ucc.asn.au/dropbear/CHANGES.

It would probably also be wise to put out an advisory to all users of the resin-os platform informing them of such a high profile security issue that could potentially affect them.

On a side note, why are you using dropbear instead of openssl? The later is much more widely audited and tested, while it seems like dropbear is full of very critical security bugs in every release they put out.

petrosagg, 2017-09-25:
Thanks for this report. This vulnerability doesn't affect resinOS since it only gets triggered when dropbear is started with the -a flag. We'll update dropbear in meta-resin anyway though.

The reason chose dropbear over openssh was because of disk and memory footprint but we're open to reconsider this. I don't have the exact numbers in mind so it might be worth to revisit

agherzan, 2018-02-01:
Update to the version which includes this fix (2017.75) will be done along with https://github.com/resin-os/meta-resin/issues/919 as yocto rocko already has this version available (http://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-core/dropbear/dropbear_2017.75.bb?h=rocko). I will close this issue though because we are not affected by the vulnerability reported as Petros pointed out above.

closed 2018-02-01 by agherzan