balena-os/meta-balena #860: Consider switching ntp client to chrony

petrosagg, 2017-09-24:
https://chrony.tuxfamily.org/comparison.html
http://layers.openembedded.org/layerindex/recipe/51402/

imrehg, 2017-09-24:
I guess triggered by this HN discussion? https://news.ycombinator.com/item?id=15324386

As a data point, been using chrony myself in Arch for quite a bit, good quality network time indeed. It does quite a bit more built in compared to `systemd-timesyncd` (with the whole server side). Would it be possible (and desirable) to have chrony without the ntp server parts, for example? (to reduce possible attack surfaces).

One possible good feature from chrony could be though, that there could be a way letting people to use GPS time to set their time (e.g. some mobile networking cases, when the 3G/4G card has GPS built in), chrony has relatively easy support for that, IMHO, so config change is enough, while not sure if possible with the current setup.

bbinet, 2017-09-27:
+1 that would be great if the time could also be set from an external GPS instead of only NTP. That would be useful for example when the device only have sporadic or occasional internet access.

willnewton, 2017-10-02:
Although it looks like timesyncd was not covered in this audit:

https://www.coreinfrastructure.org/news/blogs/2017/09/securing-network-time

agherzan, 2018-02-01:
Generally I find it as a good idea at least to switch to chrony as a starting point. This issue seems to have remained at the status of consideration and as Will pointed above, there is no comparison with timesyncd (SNTP). @petrosagg do we move forward and replace timesyncd by chrony?

PR #1077 (2018-06-12): Move from systemd-timesyncd to chrony
Fixes #860 #1068 

Tested on Raspberry Pi 3b+.

It has been considered to move to chrony as an ntp client for various reasons. This pull requests also adds a NetworkManager dispatcher script which supplies chronyd with any dhcp advertised ntp server. resinOS has been ignoring any ntp servers advertised by dhcp servers.

A comparison between chrony and systemd-timesyncd is unfair as they are different things. chrony is a full ntp implementation (client + server!). systemd-timesyncd implements a barebones sntp client only.
systemd-timesyncd has limited featu [... cut]
closed 2018-06-12 by ZubairLK