balena-os/meta-balena #867: plymouth splash daemon waiting for tty1 uses CPU resources unnecessarily.

shaunmulligan, 2017-10-04:
This plymouth process seems to be hogging CPU 
```
@sbin/plymouthd --tty=tty1 --mode=boot --pid-file=/run/plymouth/pid --attach-to-session --kernel-command-line=plymouth.ignore-serial-consoles splash
```
it looks like its consistently using 5% of the CPU on my rpi zero

willnewton, 2017-10-11:
It seems like running ```plymouth --quit`` kills the process and this is something you should do at the end of your boot process. However it is not clear when we actually get to the end of the boot and whether we want to keep the splash screen around. @agherzan any ideas what we should do here?

shaunmulligan, 2017-10-11:
perhaps the supervisor can kill plymouth just before the user application is starting up?