balena-os/meta-balena #880: Time synchronization daemon silently failes during bootup

abrodersen, 2017-10-11:
During a support request, we noticed that `systemd-timesyncd` would stop during boot up and return an exit code of `0`, causing `systemd` to not restart the service. This occured on a ResinOS 1.26 device.

[Flowdock thread](https://www.flowdock.com/app/rulemotion/user_happiness/threads/KeuwScCzTxgzh-7JLfgAJfSPsYZ)
[Front thread](https://app.frontapp.com/open/cnv_8riy5d)

Relevant logs:
````
Jan 01 00:00:04 raspberrypi3 systemd[1]: Starting Network Time Synchronization...
Jan 01 00:00:04 raspberrypi3 systemd-timesyncd[129]: System clock time unset or jumped backwards, restoring from recorded timestamp: Wed 2017-10-11 17:07:53 UTC
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Added new server pool.ntp.org.
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: systemd-timesyncd running as pid 129
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Selected server pool.ntp.org.
Oct 11 17:07:53 raspberrypi3 systemd[1]: Started Network Time Synchronization.
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Resolving pool.ntp.org...
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Failed to resolve pool.ntp.org: Temporary failure in name resolution
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Resolving pool.ntp.org...
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Failed to resolve pool.ntp.org: Temporary failure in name resolution
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Resolving pool.ntp.org...
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Failed to resolve pool.ntp.org: Temporary failure in name resolution
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Resolving pool.ntp.org...
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Failed to resolve pool.ntp.org: Temporary failure in name resolution
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Resolving pool.ntp.org...
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Failed to resolve pool.ntp.org: Temporary failure in name resolution
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Resolving pool.ntp.org...
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Failed to resolve pool.ntp.org: Temporary failure in name resolution
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Resolving pool.ntp.org...
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Failed to resolve pool.ntp.org: Temporary failure in name resolution
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Resolving pool.ntp.org...
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Failed to resolve pool.ntp.org: Temporary failure in name resolution
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Resolving pool.ntp.org...
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Failed to resolve pool.ntp.org: Temporary failure in name resolution
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Resolving pool.ntp.org...
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Failed to resolve pool.ntp.org: Temporary failure in name resolution
Oct 11 17:07:53 raspberrypi3 systemd-timesyncd[129]: Slowing down attempts to contact servers.
Oct 11 17:08:06 raspberrypi3 systemd[1]: Stopping Network Time Synchronization...
Oct 11 17:08:06 raspberrypi3 systemd[1]: Stopped Network Time Synchronization.
````

floion, 2017-10-12:
Is thia happening consistently for that user?
If so, can they also test this with the latest OS release?

pansnova, 2017-10-12:
Hello,
i'm the user that have the problem. This problem happens sometimes. Not always
Resin support tried to update the os to the latest release (2.x )but it creates a lot of problem on my application. So we decide to update at least to 1.26

willnewton, 2017-10-16:
I wonder if this issue might be relevant: https://github.com/systemd/systemd/issues/454

willnewton, 2017-10-16:
We also have an issue open for potentially switching to using chrony for time synchronization: https://github.com/resin-os/meta-resin/issues/860

agherzan, 2018-02-07:
I can't reproduce this on 2.9.6.  I tried various things to reproduce it:
1. stop dns - not have dns resolution but connect to a network - timesyncd stays up 
2. have dns up but no connectivity - timesyncd stays up

Unless somebody comes up with a specific way to reproduce, I will consider this a 1.X specific issue and recommend updating the hostOS. We are not developing on 1.X anymore.

Closing this issue but feel free to ping us if you think this is still an issue on the platform.

closed 2018-02-07 by agherzan