balena-os/meta-balena #881: DNSMasq timeouts require a sensible value

hedss, 2017-10-12:
As part of the work for OnPrem, it was noticed in situations where a domain is not resolvable, operations could timeout due to blocking. See this thread https://www.flowdock.com/app/rulemotion/r-supervisor/threads/yW_H1qeMsuuXd0y2zsM-7gliLQ2 for more information. The solution there is to alter the resolver options so that the client drops the request after a particular length of time and number of attempts.

It's entirely possible that we will suffer similar issues on resinOS via DNSMasq, including within the Supervisor.

DNSMasq itself does not have user configurable option for the timeout length of a query, but it is set in `src/config.h` as the following definitions:
```
#define TIMEOUT 10
```
DNSMasq drops queries after 4x the `TIMEOUT` value, which is hardcoded in the `get_new_frec()` function in `src/foward.c`.

Therefore by default, any lookup that does not resolve takes a full 40 seconds before it's dropped should the upstream not resolve it. This is not ideal.

A sensible value for `TIMEOUT` needs to be configured.

willnewton, 2017-10-12:
What is the problem that is trying to be solved here? Is there a testcase that reproduces the problem?

hedss, 2017-10-12:
There have been noticeable timeouts on the Supervisor (see https://github.com/resin-io/resin-api/pull/589#issuecomment-335634114), and Pablo's interested in shortening the DNS timeouts.

The other way to go with this is to alter the resolver config (shorten timeouts and attempts).

CCing in @pcarranzav for more information, and @petrosagg.

agherzan, 2018-02-01:
@pcarranzav @petrosagg @hedss What is the status of this? Do we have a conclusion on this? Do we want to shorten the timeout definition? 

Can we have a test case in order to be able to reproduce this issue?