balena-os/meta-balena #890: meta-resin check resin-boot partition free space at build time

floion, 2017-10-27:
Make meta-resin warn and error out at build time when creating the resin-boot partition if the free space in this partition is less than the biggest file which resides there (this is a requirement for being able to do atomic updates for the boot files)

floion, 2017-11-01:
Shall we do this check only for non-production builds?

agherzan, 2017-11-01:
The non production builds don't have resinhup support so given that this issue is only for atomic writes when updating the hostOS, I would say having this only for production builds would be enough.

floion, 2017-11-01:
The size of files in resin-boot can vary between development builds and production builds. This is what I meant with the above comment. That maybe we should not have this check fatal for development builds

agherzan, 2017-11-01:
What I propose is to not implement this check for development builds. Do you see a problem with this @floion ?

floion, 2017-11-01:
So we should only have the check for production builds, right?

agherzan, 2017-11-01:
Yes.

floion, 2017-11-01:
Cool, that's what I actually meant in the second comment here. Now that I look at it I see I wrote it the other way around

floion, 2017-12-04:
@agherzan this was implemented, right?

agherzan, 2017-12-04:
Implemented in https://github.com/resin-os/meta-resin/pull/895

closed 2017-12-04 by agherzan