balena-os/meta-balena #892: ssh_exchange_identification: read: Connection reset by peer

camueller, 2017-10-29:
I am aware of #387 which is already closed but the problem still exists:
```
$ ssh root@resin.local -p22222 -vvv                                                                                                                                                                        
OpenSSH_7.2p2 Ubuntu-4ubuntu2.2, OpenSSL 1.0.2g  1 Mar 2016                                                                                                                                                                                 
debug1: Reading configuration data /home/axel/.ssh/config                                                                                                                                                                                   
debug1: Reading configuration data /etc/ssh/ssh_config                                                                                                                                                                                      
debug1: /etc/ssh/ssh_config line 19: Applying options for *                                                                                                                                                                                 
debug2: resolving "resin.local" port 22222                                                                                                                                                                                                  
debug2: ssh_connect_direct: needpriv 0
debug1: Connecting to resin.local [192.168.69.76] port 22222.
debug1: Connection established.
debug1: identity file /home/axel/.ssh/id_rsa type 1
debug1: key_load_public: No such file or directory
debug1: identity file /home/axel/.ssh/id_rsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/axel/.ssh/id_dsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/axel/.ssh/id_dsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/axel/.ssh/id_ecdsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/axel/.ssh/id_ecdsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/axel/.ssh/id_ed25519 type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/axel/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_7.2p2 Ubuntu-4ubuntu2.2
ssh_exchange_identification: read: Connection reset by peer
```
After I mounted the  /dev/mmcblk0p2 on a linux machine realized that in ```/usr/bin``` the ```ssh``` command links to ```dropbearmulti```:
```
lrwxrwxrwx 1 root root 23 Aug  2 18:06 ssh -> /usr/sbin/dropbearmulti
```
After I replaced this link with a link to the real ssh ..
```
lrwxrwxrwx 1 root root 12 Okt 29 11:36 ssh -> /usr/bin/ssh
```
... I was able to use ssh without problems:
```
axel@tpw520:/data/Downloads/resin$ ssh root@resin.local -p22222
The authenticity of host '[resin.local]:22222 ([192.168.69.76]:22222)' can't be established.
RSA key fingerprint is SHA256:bN6X0D8uTNH0+xlGAQOM8Pf8CqHmuFUwzsZis7aXTzo.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '[resin.local]:22222,[192.168.69.76]:22222' (RSA) to the list of known hosts.
root@resin:~#
```

agherzan, 2017-10-30:
Hello @camueller . From the logs I see you are using resinOS unmanaged. Can you confirm? As well, what version are you reproducing this issue on?

camueller, 2017-10-30:
Unmanaged? I simply followed the instructions on https://resinos.io/docs/raspberry-pi2/gettingstarted .
This page contains the link ```https://files.resin.io/resinos/raspberry-pi2/2.2.0%2Brev1.dev/image/resin.img.zip``` I used to download the image. 
The installed image identifies as:
```
# uname -a
Linux resin 4.4.50 #2 SMP Mon Jul 31 12:36:40 UTC 2017 armv7l armv7l armv7l GNU/Linux
```

agherzan, 2017-10-30:
Got it. I'll take a look at it.

agherzan, 2018-02-01:
Hello @camueller . We updated the images and the documentation you pointed to. I was unable to reproduce this issue on this image and since the last release there are a series of fixes that could have affected this. Could you please retest with the currently documented image?

agherzan, 2018-02-05:
Closing this as not reproducible on 2.9.6.

closed 2018-02-05 by agherzan