balena-os/meta-balena #900: hostapp-update might fail to correctly update files and bail out

imrehg, 2017-11-09:
Tested on resinstaging.

Staging 2.6.0->2.7.5
```
Status: Downloaded newer image for resin/resinos:2.7.5_rev1-raspberrypi3
[INFO] Deploying /mnt/boot/start_cd.elf... done.
[INFO] Deploying /mnt/boot/bcm2709-rpi-2-b.dtb... done.
[INFO] Deploying /mnt/boot/fixup_cd.dat... done.
[INFO] Deploying /mnt/boot/system-connections/README.ignore... done.
[INFO] Deploying /mnt/boot/system-connections/resin-sample.ignore... done.
[INFO] Deploying /mnt/boot/bcm2708-rpi-0-w.dtb... done.
[INFO] Deploying /mnt/boot/bootcode.bin... done.
[INFO] Deploying /mnt/boot/resinos.fingerprint... overwriting modified file... done.
[INFO] Deploying /mnt/boot/fixup_x.dat... done.
[INFO] Deploying /mnt/boot/config.json... file modified but blacklisted. Ignoring.
[INFO] Deploying /mnt/boot/device-type.json... done.
[INFO] Deploying /mnt/boot/bcm2835-bootfiles-20170405.stamp... overwriting modified file...Error response from daemon: lstat /mnt/sysroot/inactive/docker/aufs/mnt/b318a40aba9e647325d21b087b0fd9f04ac918a6cb6ad34964fb70a2ac5d9ab3/resin-boot/bcm2835-bootfiles-20170405.stamp: no such file or directory
```

2.7.2->2.7.4+rev1

```
Status: Downloaded newer image for resin/resinos-staging:2.7.4_rev1-raspberrypi3
[INFO] Deploying /mnt/boot/start_cd.elf... done.
[INFO] Deploying /mnt/boot/start_db.elf... done.
[INFO] Deploying /mnt/boot/device-type.json... done.
[INFO] Deploying /mnt/boot/bcm2709-rpi-2-b.dtb... done.
[INFO] Deploying /mnt/boot/bcm2835-bootfiles-20170405.stamp...Error response from daemon: lstat /mnt/sysroot/inactive/docker/aufs/mnt/0523500150ec899897f51ca23b6127f78d6de7087b868c662e7b8792c624d814/resin-boot/bcm2835-bootfiles-20170405.stamp: no such file or directory
```

agherzan, 2017-11-16:
@imrehg This should be fixed by https://github.com/resin-os/meta-resin/pull/903 . Please confirm and close.

imrehg, 2017-11-20:
Fixed as above.

closed 2017-11-20 by imrehg