balena-os/meta-balena #906: Boot partition for all future OS releases should contain an os-release file

pimterry, 2017-11-16:
This should contain the same content as `/etc/os-release`.

Right now, it's very difficult to take any arbitrary OS image and tell which OS version it comes from. The CLI's best solution to that so far is to mount the root partition and read `/etc/os-release`. This is no longer easily possible with a hostapp enabled rootfs. Having the OS version file easily available in the boot partition instead will make this process much simpler for all tools that need this information.

Discussed in https://github.com/resin-io/hq/wiki/Architecture-Calls#15-nov-2017 (as a tangent in the 'image reconfiguration bundle' topic)

pimterry, 2018-01-18:
@agherzan did we make any progress on this? It would be _really_ useful to have it in ASAP, it'd be a huge help long-term to have this in place before multicontainer, so we can clearly recognize multicontainer images externally in future.

agherzan, 2018-01-22:
@pimterry I'm going to work on this this week. Added to my queue.

closed 2018-01-26 by agherzan

pimterry, 2018-01-29:
Coming back from holiday to have this fixed & shipped already is really great, thank you @agherzan!

agherzan, 2018-01-29:
@pimterry I told you vacations are overrated.