balena-os/meta-balena #919: meta-resin support for Yocto Rocko

floion, 2017-12-04:


floion, 2018-01-15:
Any update on this?

agherzan, 2018-01-22:
Hi @floion . There was no work done towards this issue hence no update. PRs will be linked when progress is done.

agherzan, 2018-02-14:
I will assign Florin as I know he has started work on this.

floion, 2018-02-15:
Removed myself from assigned.
Did the following PRs on this:
https://github.com/resin-os/meta-resin/pull/993
https://github.com/resin-os/meta-resin/pull/992
https://github.com/resin-os/meta-resin/pull/988
https://github.com/resin-os/meta-resin/pull/987
https://github.com/resin-os/meta-resin/pull/986
https://github.com/resin-os/meta-resin/pull/985
https://github.com/resin-os/meta-resin/pull/984
https://github.com/resin-os/meta-resin/pull/983
https://github.com/resin-os/meta-resin/pull/982
https://github.com/resin-os/meta-resin/pull/981

In progress I left: https://github.com/resin-os/meta-resin/pull/991

Support is not yet done. still a few changes are still needed

KiaFathi, 2018-02-22:
Any update on this & #991 since last Thursday?

agherzan, 2018-02-22:
We are working on it and we will update as soon as we have progress.

agherzan, 2018-02-27:
The rocko support is not part of 2.10.0 meta-resin release. Closing this issue.

closed 2018-02-27 by agherzan