balena-os/meta-balena #935: Upgrade from 2.0.3 to 2.9.1

Bnaik, 2017-12-28:
I have tried upgrading resinos from 2.0.3 to 2.9.1, and it worked well. But after upgrade It says galena-host service failed. because resin-inactive.mount service failed.
Any help?
And it still shows filesystem ext4. How it can be changed to aufs?

agherzan, 2017-12-28:
Hi @Bnaik . This is an issue that we encountered in the past and fixed at this time. I'll ping @imrehg who knows more about it and will take a look.

imrehg, 2018-01-01:
@Bnaik what tool have you used for the update? resinHUP needs some changes for the balena versions (as 2.9.1 is, for example), here's the pending issue: https://github.com/resin-os/resinhup/issues/120 . What hardware are you using as well?

Bnaik, 2018-01-02:
@imrehg We have upgraded using this script: https://github.com/resin-os/resinhup/blob/master/upgrade-2.x.sh (For version 2.0.3)
We are using raspberryPi3. Any manual way to upgrade from docker to balena?

imrehg, 2018-01-05:
@Bnaik that is being worked on right now, should come very soon.

agherzan, 2018-02-01:
@imrehg This something we can close? As far as I know this was handled. Otherwise can we move this ticket as this doesn't seem to be a meta-resin issue.

imrehg, 2018-02-03:
@Bnaik the latest `resinhup` script should now handle balena migration, and can give it a try. If there are any problems, please open an issue on that repo. Thanks!

closed 2018-02-03 by imrehg