balena-os/meta-balena #938: plymouth splash daemon uses tty1 causing problems with users trying to use keyboard input

pcarranzav, 2018-01-03:
Closely related to #867 - the running plymouth daemon makes keyboard input not work properly on a console in the user's container.
Something should probably call `plymouth quit` or start the `plymouth-quit.service` at the end of the boot process?

agherzan, 2018-02-07:
Closing this as a duplicate of https://github.com/resin-os/meta-resin/issues/867

ZubairLK, 2019-09-27:
Noticed this again. On a nuc image running in a VM. The display console is almost unusable. keyboard strokes are dropped randomly. `plymouth quit` makes the issue go away

agherzan, 2019-10-02:
I hit this lately as well. It sounds like a regression. It used to work just fine a couple of weeks ago.

ZubairLK, 2019-10-02:
@agherzan I've pretty much noticed this since I joined on the nuc vm..

agherzan, 2019-10-02:
Might have been correlated with a VirtualBox update.

rgilio, 2019-10-10:
I've also been running into this issue and can confirm running ''plymouth quit'' makes the issue go away. Also building BalenaOS without ''plymouth'' corrects the behavior.

ZubairLK, 2019-10-29:
So there is a 'feature' in plymouth that is causing this I think.
From https://cgit.freedesktop.org/plymouth/tree/docs/development.txt
```
Keyboard commands
~~~~~~~~~~~~~~~~~

plymouthd can be controlled with the keyboard. Use the following keys:

 * +Esc+: Toggle between system console and plymouth boot animation.

 * +Ctrl-V+: Toogle verbose mode on and off.

 * +Ctrl-U+ or +Ctrl-W+: erase a line
```

Plymouth is intercepting keyboard inputs.

Option a) Ask plymouth to ignore keyboard input (if plymouth supports that)
Option b) Add a udev rule that prevents /dev/input going to the plymouthd process