balena-os/meta-balena #945: Allow users to set up bridges through NetworkManager.

abrodersen, 2018-01-08:
Requires removing `type:bridge` from the `unmanaged-devices` field in `/etc/NetworkManager/NetworkManager.conf`.

HQ Issue: https://github.com/resin-io/hq/issues/1142
Flowdock Thread: https://www.flowdock.com/app/rulemotion/user_happiness/threads/f9qNRn1MKXw2kGYRWXn6nCkYTXA

abrodersen, 2018-01-09:
Accoring to Andrei, the exception of bridges from management was added to prevent Network Manager from stomping on Docker's bridge device. It appears we can use replace `type:bridge` with `interface-name:docker*` to allow user bridges and still prevent Network Manager from interfering with Docker.

agherzan, 2018-01-09:
Fixed by: https://github.com/resin-os/meta-resin/pull/949

closed 2018-01-10 by agherzan