balena-os/meta-balena #946: Investigate boot time on resinOS

agherzan, 2018-01-08:
Measure and graph the current boot time on Raspberry 1 (0, or 0 wifi) and compare it to current latest version of Raspbian. Provide graphs for both of the systems.

agherzan, 2018-02-06:
Results were provided in: https://docs.google.com/spreadsheets/d/1b9g8iMivhJN4SonzHLcqS8Ol3RNC6dl5v8oGber108k/edit#gid=0 . Based on this, OS team will continue the task of optimizing the boot time.

pebbie, 2018-11-20:
> Results were provided in: https://docs.google.com/spreadsheets/d/1b9g8iMivhJN4SonzHLcqS8Ol3RNC6dl5v8oGber108k/edit#gid=0 . Based on this, OS team will continue the task of optimizing the boot time.

can you make the link available for public (read-only)?

RomanBapst, 2019-01-29:
@agherzan Could you please share the results of the upper document? I would like to see how balena performs compared to raspbian.

agherzan, 2019-01-30:
That investigation is outdated now and we will resume this as part of our next activities.

ZubairLK, 2019-02-25:
quick check, https://github.com/balena-os/meta-balena/pull/1408 will shave of roughly 5 seconds

There is some room for improvement by reordering services/delaying services. But it needs to be done carefully.
On a pi0, disabling almost all services except balena resulted in 
```
root@localhost:~# systemd-analyze 
Startup finished in 8.451s (kernel) + 30.250s (userspace) = 38.702s
multi-user.target reached after 30.058s in userspace
root@localhost:~# 
```

There might be some applications that might not need VPN/NetworkManager/ModemManager/resin-supervisor to be online before their application which could benefit from such an improvement. e.g. A configurable delay in selected services could improve the first application response time.

But implementation details aside, the easiest ones are `balena-host.service` `mnt-sysroot-inactive.service`

ZubairLK, 2019-02-26:
note: scripts in initramfs and filesystem mount such as resin-partition-mounter have `sleep 1`. Might be worth checking with `sleep 0.1`. All other services block on the partition being mounted. This might shave a second or two.

ZubairLK, 2019-03-11:
Adding some notes while looking into upx.
As of https://github.com/balena-os/meta-balena/commit/69bc7e1c6c8382d69b57b662518909c4f7824644, we switched to lzma by default which offers better compression at the cost of higher cpu usage. I checked the impact of upx on the binaries we compress to get some idea. Currently we compress `os-config` and `mobynit`. Both do get used during boot.

```
root@62d0b9a:/mnt/data# time os-config-decompress -V
os-config 1.0.0
real    0m0.067s
user    0m0.029s
sys     0m0.031s
# Default algorithm NRV2E
root@62d0b9a:/mnt/data# time os-config_upxed_best -V
os-config 1.0.0
real    0m0.216s
user    0m0.130s
sys     0m0.058s
# LZMA compression
root@62d0b9a:/mnt/data# time os-config_upxed_lzma_best -V
os-config 1.0.0
real    0m0.448s
user    0m0.370s
sys     0m0.043s
```

ZubairLK, 2019-11-06:
On a pi0 , boot time rose from ~30s in 2.32 to ~46s in 2.44. Needs investigating as to why this happened.

robertgzr, 2019-11-15:
@ZubairLK this user sees a pretty high impact on boot time on pi0 / 2.44 going up to ~6min, is their information any useful

see https://github.com/balena-os/balena-engine/issues/105#issuecomment-550281934