balena-os/meta-balena #947: openvpn preparation script hardcodes the vpn port

floion, 2018-01-09:
https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-connectivity/openvpn/openvpn/prepare-openvpn#L7

This should better be read from config.json instead

floion, 2018-01-09:
The link I posted above shows the vpn port is 443, so hardcoded in the prepare-openvpn script: https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-connectivity/openvpn/openvpn/prepare-openvpn#L7

Suggestion was that prepare openvpn should fetch the vpn port from the config.json instead where the vpn port is declared.

agherzan, 2018-01-11:
I understood that. My question is why do we want that? Is it a configurable variable now? Do we plan to change 443?

floion, 2018-01-11:
Since we have this variable in config.json it seems natural to fetch it from there and not have it hardcoded.
Just like it is done for the vpnEndpoint. Otherwise I see no point in declaring the same variable in multiple places

agherzan, 2018-01-11:
Now I understand. I was not aware this was introduced in the json. Most probably when that script was implemented that was not the case. Please update the description with the fact that this is requested because the json has the vpn port as well.

agherzan, 2018-03-08:
This will be reverted for now until the CLI gets fixed: https://github.com/resin-os/meta-resin/pull/1020 . CC @jviotti @zvin

agherzan, 2018-03-08:
The above mentioned PR was changed to only default if 1723 is found in the config. So we can close this issue.

commit 7fbf739f (2018-02-02): openvpn: Use VPN_PORT from resin-vars in prepare-openvpn

Fixes #947

Change-type: patch
Changelog-entry: Use VPN port from config.json (defaulting it to 443)
closed 2018-03-08 by agherzan