balena-os/meta-balena #953: Segfault when running on VirtualBox

ruudboon, 2018-01-13:
I'm trying to run ResinOs (intel, resinOS 2.2.0+rev1.dev) on VirtualBox with the following settings

- Disabled I/O APIC
- Enabled EFI
-1024MB base memory

I see the bootloader and select flash. System boots and see some network setup is being done, after that I get a segmentation fault. Are there any settings that I'm missing?

[image]

floion, 2018-01-13:
Hi. Are you running the qemu image?

ruudboon, 2018-01-13:
Hi, No. Just the Oracle Virtual box on a MacBook. Looked at [this](https://resin.io/blog/no-hardware-use-virtualbox/) tutorial for the settings

ruudboon, 2018-01-13:
I tried the image of ResinIO but seeing the same issue with version 2.2.0.
With resin-Test-2.7.8+rev1-dev-v6.4.2 I see a different message:
[image]

floion, 2018-01-13:
Thanks for the report. We'll take a look and come back to you

ruudboon, 2018-01-13:
@floion Thnx. Please let me know if you need more info.

imrehg, 2018-01-19:
@ruudboon I've checked out 2.2.0, 2.7.8, 2.9.6 versions of resinOS, and all of them provision & function just fine on VirtualBox 5.2.6 that I have at hand.

Can you please check that the settings are indeed the same as in the blog? For example that you are using SATA disks? convert the image to `.vmdk` format? the chipset is `PIIX3`? So far I cannot reproduce the issues you encountered.....

(cc @floion )

ruudboon, 2018-01-19:
@imrehg All the settings mentioned on the blog are the same. Maybe it's something OSX related?
Running VirtualBox Version 5.1.30 r118389 (Qt5.6.3), updating now.

ruudboon, 2018-01-19:
@imrehg Unfortunately same issue with latest version.

imrehg, 2018-01-19:
I'm running this on Linux. It's possible the OSX is a problem (well, rather VirtualBox on OSX). In general, running virtual devices we'd recommend the QEMU device type, that is created for virtualization in the first place. We'll check whether we can reproduce the issue on OSX, and would you want to check out the QEMU type?

ruudboon, 2018-01-19:
@imrehg Offcourse, will give it a shot this weekend. Got some more Resin thing I would like to try :D

imrehg, 2018-01-25:
@ruudboon have you had a chance to try this out? 
We've tried to replicate it with resinOS 2.9.6+rev1, using VirtualBox 5.2.6 on MacOS 10.12 that we have at hand, and everything works just fine, as much as I can tell (the "virtual" device flashes, provisions, comes online, downloads and runs the user application)

What other settings you might have changed (e.g. NAT)? If there were any other settings changed, can try it with a vanilla setup as in the blogpost (just the "motherboard" settings and the SATA storage driver). Just trying to see the differences.

imrehg, 2018-02-03:
@ruudboon just wondering if there's any update on this issue? Thanks!

ruudboon, 2018-02-06:
@imrehg Did some more testing, changing settings like removing floppy, optical etc but keep having the segfault. I can drop the the virtual box if you want?

imrehg, 2018-02-06:
@ruudboon it's more about we don't have enough info so far about what exactly have you tried

* What OSX version you are running?
* Have you tried the latest VirtualBox (5.2.6, like mentioned above)?
* Which networking setting do you use?

The floppy/optical should not matter at all, IMHO.

ruudboon, 2018-02-06:
Running High Sierra (10.13.2)
Running latest VirtualBox 5.2.6 
Running standard network setting with attached to nat, changed it to bridged but with the same result.

imrehg, 2018-02-06:
@ruudboon got hold of a High Sierra (10.13.3), using VirtualBox 5.2.6, latest resinOS 2.9.7+rev1 for Intel NUC, and all flashing / running the image works fine here. The only things I needed to set are the System/Motherboard:
[image]
and Storage:
[image]
Everything else kept as default.

It flashes just fine. You see any difference compared to your setup? Running out ideas of what can be different between the working and non-working case.

ruudboon, 2018-02-08:
Weird stuff... I will try to redownload the image tomorrow. Maybe it's corrupted somehow...??

[image]
[image]
[image]

agherzan, 2018-03-20:
Any updates on this? I would like to close the issue as it doesn't see me to be meta-resin, resinOS core issue.

ruudboon, 2018-03-20:
@agherzan No, but if i'm the only one with this problem I think we should close it. If I can find something that is reproducible on a different computer I will reopen this.

ColPanic, 2018-09-26:
You are not the only one having this issue.  I'm having the exact same.  It gets to the login screen, then after a few seconds says no floppy found and segfaults. driving me bonkers.

Anybody fix it?

agherzan, 2018-10-19:
@ColPanic Any updates on this? Can you share information about a setup we can use to reproduce?

ZubairLK, 2020-01-27:
Closing stale issues for cleanup. Please reopen if needed.

closed 2020-01-27 by ZubairLK