balena-os/meta-balena #958: User-defined bridging needs custom iptables FORWARD rules for IPV4 traffic

lekkas, 2018-01-26:
This is an issue a resinOS user hit recently on `v2.9.6+rev1(prod)`. It looks like a fairly common use-case (IPV4 bridging of two ethernet interfaces), so we might want to consider adapting the default behaviour so that users don't have to add custom iptable rules.

The custom iptables rules the user had to add were

```
# iptables -A FORWARD -i cb0 -j ACCEPT
# iptables -A FORWARD -o cb0 -j ACCEPT
```

See: https://forums.resin.io/t/bridged-network-configuration/2325/13 (public discourse message with more details on the solution and the problem)
See: https://www.flowdock.com/app/rulemotion/public-s-community/threads/7fkcHbuqAMCc440iCA-GRB_Pung

ZubairLK, 2018-04-30:
I don't see the bug. Am I missing something here?
I ran this on a pi3 Resin OS 2.12.5+rev2 (dev)

```
root@ba6d524:~# iptables -S
-P INPUT ACCEPT
-P FORWARD ACCEPT
-P OUTPUT ACCEPT
-N DOCKER
-N DOCKER-ISOLATION
-N DOCKER-USER
... many other rules ...
root@ba6d524:~#
```

Shouldn't the '-P FORWARD ACCEPT' rule be applied and therefore, in the case of bridging network interfaces, the interface specific rules shouldn't need to be created.

I'll run iptables -S on a v2.9.6 image and see what happens

ZubairLK, 2018-05-09:
I think this might have been an issue at some point. However, I cannot reproduce it in current dev images. There is no need to set iptable rules to set up a bridge.

For reference, I'll put the info here.

My setup
Beaglebone Black (BBB) with a USB Ethernet dongle.
eth0 is BBB ethernet port
eth1 is BBB USB Ethernet dongle

eth0 is connected to internet
eth1 is connected to PI3

Pi 3 B plus
eth0 is connected to eth1 on BBB via USB Ethernet dongle

The plan is to bridge eth0 and eth1 on BBB so that eth0 on pi3 can receive a dhcp lease and connect to the internet

```
root@ab6c3f3:~# cat /etc/os-release
ID="resin-os"
NAME="Resin OS"
VERSION="2.12.3+rev1"
VERSION_ID="2.12.3"
PRETTY_NAME="Resin OS 2.12.3+rev1"
RESIN_BOARD_REV="9f222a8"
META_RESIN_REV="c286ca0"
SLUG="beaglebone-black"
MACHINE="beaglebone"
VARIANT="Development"
VARIANT_ID="dev"
root@ab6c3f3:~#
```

Setting up the network bridge (derived from https://people.freedesktop.org/~lkundrak/nm-docs/nmcli-examples.html)
On BBB, these commands set up the bridge
```
nmcli con down "Wired connection 1"
nmcli con down "Wired connection 2"
nmcli con add type bridge con-name cb0 ifname cb0
nmcli con down "cb0"
nmcli con add type ethernet con-name br-slave-1 ifname eth0 master cb0
nmcli con add type ethernet con-name br-slave-2 ifname eth1 master cb0
nmcli con modify cb0 bridge.stp no
nmcli con show
nmcli con up "cb0"
```

Some console output on BBB
```
root@ab6c3f3:~# nmcli con show
NAME                UUID                                  TYPE      DEVICE
br-slave-1          83121ae2-0099-435c-8e8f-7c972d154aa0  ethernet  eth0
br-slave-2          b90218a8-ad72-4d8f-9f88-3bc637da0668  ethernet  eth1
cb0                 0d620267-4309-4b14-8bd7-78171ce50a03  bridge    cb0
resin-dns           494c4557-4ebe-4812-a76b-3ec1bd9f0c82  bridge    resin-dns
supervisor0         c2d3a66a-8c6a-4fef-943c-27e863bf62c8  bridge    supervisor0
Wired connection 1  2fb3f00a-6604-34a7-b933-64c7a201bc9f  ethernet  --
Wired connection 2  440cd6d6-83b6-3a61-8f92-184552240175  ethernet  --
root@ab6c3f3:~# cat /etc/NetworkManager/system-connections/
README.ignore        br-slave-2           resin-sample.ignore
br-slave-1           cb0
root@ab6c3f3:~# cat /etc/NetworkManager/system-connections/cb0
[connection]
id=cb0
uuid=0d620267-4309-4b14-8bd7-78171ce50a03
type=bridge
interface-name=cb0
permissions=

[bridge]
stp=false

[ipv4]
dns-search=
method=auto

[ipv6]
addr-gen-mode=stable-privacy
dns-search=
method=auto
root@ab6c3f3:~# cat /etc/NetworkManager/system-connections/br-slave-1
[connection]
id=br-slave-1
uuid=83121ae2-0099-435c-8e8f-7c972d154aa0
type=ethernet
interface-name=eth0
master=cb0
permissions=
slave-type=bridge

[ethernet]
mac-address-blacklist=
root@ab6c3f3:~# cat /etc/NetworkManager/system-connections/br-slave-2
[connection]
id=br-slave-2
uuid=b90218a8-ad72-4d8f-9f88-3bc637da0668
type=ethernet
interface-name=eth1
master=cb0
permissions=
slave-type=bridge

[ethernet]
mac-address-blacklist=
root@ab6c3f3:~# iptables -S
-P INPUT ACCEPT
-P FORWARD ACCEPT
-P OUTPUT ACCEPT
-N DOCKER
-N DOCKER-ISOLATION
-N DOCKER-USER
-A INPUT -i supervisor0 -p tcp -m tcp --dport 48484 -j ACCEPT
-A INPUT -i lo -p tcp -m tcp --dport 48484 -j ACCEPT
-A INPUT -i docker0 -p tcp -m tcp --dport 48484 -j ACCEPT
-A INPUT -i tun0 -p tcp -m tcp --dport 48484 -j ACCEPT
-A INPUT -i resin-vpn -p tcp -m tcp --dport 48484 -j ACCEPT
-A INPUT -p tcp -m tcp --dport 48484 -j REJECT --reject-with icmp-port-unreachable
-A FORWARD -j DOCKER-USER
-A FORWARD -j DOCKER-ISOLATION
-A FORWARD -o balena0 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -o balena0 -j DOCKER
-A FORWARD -i balena0 ! -o balena0 -j ACCEPT
-A FORWARD -i balena0 -o balena0 -j ACCEPT
-A FORWARD -o supervisor0 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -o supervisor0 -j DOCKER
-A FORWARD -i supervisor0 ! -o supervisor0 -j ACCEPT
-A FORWARD -i supervisor0 -o supervisor0 -j ACCEPT
-A FORWARD -o br-83eb19cd35c9 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -o br-83eb19cd35c9 -j DOCKER
-A FORWARD -i br-83eb19cd35c9 ! -o br-83eb19cd35c9 -j ACCEPT
-A FORWARD -i br-83eb19cd35c9 -o br-83eb19cd35c9 -j ACCEPT
-A DOCKER-ISOLATION -i br-83eb19cd35c9 -o balena0 -j DROP
-A DOCKER-ISOLATION -i balena0 -o br-83eb19cd35c9 -j DROP
-A DOCKER-ISOLATION -i supervisor0 -o balena0 -j DROP
-A DOCKER-ISOLATION -i balena0 -o supervisor0 -j DROP
-A DOCKER-ISOLATION -i br-83eb19cd35c9 -o supervisor0 -j DROP
-A DOCKER-ISOLATION -i supervisor0 -o br-83eb19cd35c9 -j DROP
-A DOCKER-ISOLATION -j RETURN
-A DOCKER-USER -j RETURN
root@ab6c3f3:~#
```

On pi3
```
root@565be14:~# dhclient eth0
ip: RTNETLINK answers: File exists
mv: inter-device move failed: '/tmp/resolv.conf.dhclient-new' to '/etc/resolv.conf'; unable to remove target: Read-only file system
root@565be14:~# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr FE:7B:30:4F:39:7E
          inet addr:192.168.0.18  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::36ce:3750:b925:11f9%1996087296/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:137 errors:0 dropped:0 overruns:0 frame:0
          TX packets:272 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:37195 (36.3 KiB)  TX bytes:52078 (50.8 KiB)
root@565be14:~# cat /etc/os-release
ID="resin-os"
NAME="Resin OS"
VERSION="2.12.6+rev1"
VERSION_ID="2.12.6"
PRETTY_NAME="Resin OS 2.12.6+rev1"
RESIN_BOARD_REV="c1ac9a0"
META_RESIN_REV="b744564"
SLUG="raspberrypi3"
MACHINE="raspberrypi3"
VARIANT="Development"
VARIANT_ID="dev"
root@565be14:~#
```

I'll do a quick test on production images and double confirm.

ZubairLK, 2018-05-09:
Works in production images too.
A bit tricky to test as you can't easily bring down Ethernet ports over serial getty by default on production images. So basically had to set up the bridge and run the up/down parts in one line.
```
nmcli con down "Wired connection 1" && nmcli con down "Wired connection 2" && nmcli con up "cb0"
```

closed 2018-05-09 by ZubairLK