balena-os/meta-balena #970: Screenly Sleeping not showing asset

lawlortaxis, 2018-02-05:
Hi

I am new to coding and I hope this is not a silly question. 

When Screenly should be showing the image/website it is sleeping. no matter how long I set the image for it will sleep for that long.

> 05.02.18 09:33:32 (+0000) Feb 05 09:33:32 a1fc6d8 python[226]: Sleeping for 10
> 05.02.18 09:33:42 (+0000) Feb 05 09:33:42 a1fc6d8 python[226]: Showing asset wallpaper_LCS.jpg (image)
> 05.02.18 09:33:42 (+0000) Feb 05 09:33:42 a1fc6d8 python[226]: Sleeping for 10
> 05.02.18 09:33:52 (+0000) Feb 05 09:33:52 a1fc6d8 python[226]: Showing asset wallpaper_LCS.jpg (image)
> 05.02.18 09:33:52 (+0000) Feb 05 09:33:52 a1fc6d8 python[226]: Sleeping for 10
> 05.02.18 09:34:02 (+0000) Feb 05 09:34:02 a1fc6d8 python[226]: Generating asset-list...
> 05.02.18 09:34:02 (+0000) Feb 05 09:34:02 a1fc6d8 python[226]: Showing asset wallpaper_LCS.jpg (image)
> 05.02.18 09:34:02 (+0000) Feb 05 09:34:02 a1fc6d8 python[226]: Sleeping for 2
> 05.02.18 09:34:04 (+0000) Feb 05 09:34:04 a1fc6d8 python[226]: Showing asset wallpaper_LCS.jpg (image)
> 05.02.18 09:34:04 (+0000) Feb 05 09:34:04 a1fc6d8 python[226]: Sleeping for 2
> 05.02.18 09:34:06 (+0000) Feb 05 09:34:06 a1fc6d8 python[226]: Showing asset wallpaper_LCS.jpg (image)
> 05.02.18 09:34:06 (+0000) Feb 05 09:34:06 a1fc6d8 python[226]: Sleeping for 2

imrehg, 2018-02-05:
Hi @lawlortaxis, are you using Screenly OSE? If so, would recommend opening an issue there, as it should be a question for the Screenly application https://github.com/Screenly/screenly-ose

agherzan, 2018-02-05:
This indeed doesn't sound like a core os issue. I'll close it for now but feel free to comment back if you find that this doesn't work due to a malfunction of the resinOS.

closed 2018-02-05 by agherzan