balena-os/meta-balena #1001: Docker command not found on resinos Raspberry pi 3

mohamed-elsabagh, 2018-02-21:
I am following your [guide](https://resinos.io/docs/raspberrypi3/gettingstarted/) here, and I managed to flash resinos to my sd card and ssh to it but when I run this command ```docker info``` I get ```-sh: docker: command not found```
Do I have to install docker myself, and how can I do this on resinos??

imrehg, 2018-02-21:
@mohamed-elsabagh which version of resinOS are you using? On 2.9.0 and newer, `docker` is replaced by `balena` (see more info in our blogpost at https://resin.io/blog/announcing-balena-a-moby-based-container-engine-for-iot/ ).

The documentation needs to be updated, and the relevant repository for that is https://github.com/resin-os/resinos

closed 2018-02-21 by imrehg

agherzan, 2018-02-21:
The documentation was already updated in https://github.com/resin-os/resinos/pull/448/files and you will be able to see this change in the website very soon.

mohamed-elsabagh, 2018-02-21:
Thanks so much that was very fast