balena-os/meta-balena #1005: Enable `--experimental` flag for bluez startup

mccollam, 2018-02-22:
Support for some BLE devices in bluez is hidden behind the `--experimental` flag for that service.  Enabling support for those devices simply requires the flag to be added to the systemd service definition as outlined here: https://learn.adafruit.com/install-bluez-on-the-raspberry-pi/installation#enable-bluetooth-low-energy-features

This should have no impact on other devices, so appears to be safe to do.

brynmathias, 2018-02-22:
Just for reference I'm not sure that testing the adafuit python libs with this version of bluez (its too recent) will work as they're a bit out of date, but I do have some test code that works on raspberrian for connecting to the device.

I'll see if my custom host os works and if so give you a PR for the change.

brynmathias, 2018-02-22:
ok, I only edited the init.d script for bluetoothd, not the systemd one (which is installed and used by bluez)
I might need a bit of a walk though for how to add a recipe for custom systemd scripts

brynmathias, 2018-02-23:
updates:

In my build dir I have
`puddlejumper git/resin-raspberrypi ‹master*› » cat ./build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/rootfs/etc/systemd/system/bluetooth.target.wants/bluetooth.service                      1 ↵
[Unit]
Description=Bluetooth service
Documentation=man:bluetoothd(8)
ConditionPathIsDirectory=/sys/class/bluetooth

[Service]
Type=dbus
BusName=org.bluez
ExecStart=/usr/lib/bluetooth/bluetoothd -E
NotifyAccess=main
#WatchdogSec=10
#Restart=on-failure
CapabilityBoundingSet=CAP_NET_ADMIN CAP_NET_BIND_SERVICE
LimitNPROC=1
ProtectHome=true
ProtectSystem=full

[Install]
WantedBy=bluetooth.target
Alias=dbus-org.bluez.service
`

which has the experimental flag set, but the systemd service on the host os I have built doesn't have the flag, so there's something I'm doing wrong.

agherzan, 2018-02-26:
@brynmathias I pushed a PR (https://github.com/resin-os/meta-resin/pull/1008) which should add the experimental flag to bluetoothd. Are you able to give it a try? You can build any resin-* repository but before triggering the build make sure the submodule is checked out to the branch on which this PR is made on.

brynmathias, 2018-02-26:
Trying a build now

commit caffd438 (2018-02-26): bluez5: Run bluetoothd with experimental flag

Fixes #1005

Currently features like (BLE support) are only available when running the
daemon with experimental flag.

Change-type: patch
Changelog-entry: Run bluetoothd with --experimental flag
commit 86dbf351 (2018-02-27): bluez5: Run bluetoothd with experimental flag

Fixes #1005

Currently features like (BLE support) are only available when running the
daemon with experimental flag.

Change-type: patch
Changelog-entry: Run bluetoothd with --experimental flag
closed 2018-02-27 by agherzan