balena-os/meta-balena #1051: port forwarding broke between 2.7.8 and 2.9.6

Hexagenic, 2018-04-04:
On a project I'm working we have the use case that we want to be able to access the router a device is running on, in order to grab some information about the network it's running on.

The router is not accessibly from WAN, we have to access it through a custom OpenVPN connection that every device has back to our backend servers. We've currently set it up so that every resin device forwards port 8443 on the vpn, to 443 on its router using the following code which is run every time it connects to OpenVPN through the `up` command.

```bash
TUNNEL_DEVICE=$1
TUNNEL_MTU=$2
LINK_MTU=$3
LOCAL_IP=$4
REMOTE_IP=$5
ACTION=$6
ROUTER_IP=$(ip route show default | grep default | cut -d' ' -f3)

# Remove old rules if any
iptables-save | grep -v 'custom-rule-' | iptables-restore

iptables -t nat -A PREROUTING  -i tun0 -p tcp --dport 8443 -j DNAT --to-destination ${ROUTER_IP}:443 -m comment --comment "custom-rule-dnat"
iptables -t nat -A POSTROUTING -o wlan0 -p tcp -j MASQUERADE -m comment --comment "custom-rule-masquerade"
iptables -A FORWARD -i tun0  -o wlan0 -p tcp -d ${ROUTER_IP} --dport 443 -j ACCEPT -m comment --comment "custom-rule-forward"
iptables -A FORWARD -i wlan0 -o tun0  -p tcp -m state --state RELATED,ESTABLISHED -j ACCEPT -m comment --comment "custom-rule-backward"
```

This works great on the `2.7.8` release but has stopped working on `2.9.6` and I can't figure out why, the connection is simply timing out.

This seems similar #958

Hexagenic, 2018-04-06:
The problem was unrelated to the version of the Resin image. Turns out our router had decided to blacklist that particular unit.

Sorry to bother you :)

closed 2018-04-06 by Hexagenic

agherzan, 2018-04-06:
No worries @Hexagenic . Glad you managed to find the roots of the issue.