balena-os/meta-balena #1058: How to secure a dev image during local development?

kwmlodozeniec, 2018-04-23:
I get that open ssh is great when you are developing, but I would still like to secure it with a password so other devs cannot push stuff onto my RPi while I am tinkering. Anyway this can be achieved?

agherzan, 2018-04-23:
Hello @kwmlodozeniec currently we don't have a way to do that easily. Obviously, there is always a way to run custom builds with needed configuration for the root user's password (see: https://wiki.yoctoproject.org/wiki/FAQ:How_do_I_set_or_change_the_root_password).

kwmlodozeniec, 2018-04-24:
Right, thanks for that, I will take a look. Ideally, I would not want to be building my own images just to get a secure development setup but I will give it a go. Any chance this could be turned into a feature request for future releases?

bleggett, 2018-04-26:
Allowing people to drop a public key into the image and only supporting key auth with SSH would be the best of both worlds. No open SSH, no password entry needed.

agherzan, 2018-10-19:
We've made a big step towards fixing this by adding custom ssh public keys support in `config.json` as of https://github.com/resin-os/meta-resin/pull/1227. The last step will be to configure dropbear to use only ssh keys. Maybe that done configurable. CC @shaunmulligan

commit 16a5002e (2021-07-15): README: Remove image variants and document development mode

The dual image variants are replaced with a runtime configuration switch
for development mode.

Fixes #2226
Fixes #1058
Relates to: balena-io/balena-io#2247

Change-type: minor
Changelog-entry: Replace image variants with development mode
commit 175d737e (2021-08-03): README: Remove image variants and document development mode

The dual image variants are replaced with a runtime configuration switch
for development mode.

Fixes #2226
Fixes #1058
Relates to: balena-io/balena-io#2247

Change-type: minor
Changelog-entry: Replace image variants with development mode
commit e0c150ac (2021-08-09): README: Remove image variants and document development mode

The dual image variants are replaced with a runtime configuration switch
for development mode.

Fixes #2226
Fixes #1058
Relates to: balena-io/balena-io#2247

Change-type: minor
Changelog-entry: Replace image variants with development mode
commit 6f1b09c5 (2021-09-14): README: Remove image variants and document development mode

The dual image variants are replaced with a runtime configuration switch
for development mode.

Fixes #2226
Fixes #1058
Relates to: balena-io/balena-io#2247

Change-type: minor
Changelog-entry: Replace image variants with development mode
closed 2021-09-15 by alexgg