balena-os/meta-balena #1062: docker-disk will not build with a remote docker server

mark-grimes, 2018-04-27:
There are several use cases for using a "remote" docker server, in particular when running `barys` itself from a docker container.  Yocto supports building inside a docker container through [CROPS](https://github.com/crops/docker-win-mac-docs/wiki).  `docker-disk` wants to start docker containers which complicates things, but this can be achieved through various means e.g. [docker-in-docker](https://hub.docker.com/_/docker/) (which I think is what `docker-disk` actually does?).

However, this requires a "remote" docker (remote to the docker client, even though the daemon is technically on the same machine). `docker-disk` performs a check that docker is available (in [meta-resin-common/recipes-containers/docker-disk/docker-utils.inc](https://github.com/resin-os/meta-resin/blob/v2.12.5/meta-resin-common/recipes-containers/docker-disk/docker-utils.inc#L20)) but this includes a check for a `/var/run/docker.pid` file so that even a working docker will fail if it uses a remote server.

If I hard code the `usable_docker()` function to always return "yes" I can get resin to start building from inside the [crops/poky](https://hub.docker.com/r/crops/poky/) container (note that a bit more work is required, e.g. install npm, which I won't go into here).  The function does also check by just trying to use docker at [line 36](https://github.com/resin-os/meta-resin/blob/v2.12.5/meta-resin-common/recipes-containers/docker-disk/docker-utils.inc#L36), so maybe the preceding pid check could just be removed?

agherzan, 2018-04-27:
Hi @mark-grimes . We actually use in docker builds in our build pipeline at resin. See: https://github.com/resin-os/resin-yocto-scripts/blob/master/automation/prepare-and-start.sh . I'm not familiar with crops, but you can use the Dockerfile in the above mentioned directory and build resin images using it.

closed 2018-05-14 by agherzan

mark-grimes, 2018-05-15:
The docker image mentioned does work - pretty much the only difference is that the docker daemon is run in the background on the same container, whereas when using crops I was using it in a separate network linked container.  As mentioned I think the pid check is unnecessary and limits how resin images can be built.  If I test and put in a PR do you want the contribution?

There is also a documentation issue - the Resin OS docs say to use the Yocto docs to create the required environment, and these all talk about building with crops if using containers.  Since this doesn't work (at least not without extra steps) is it worth adding @agherzan's information above somewhere to the documentation?