balena-os/meta-balena #1063: Add teensy udev rules

hpgmiskin, 2018-04-30:
Without Teensy udev rules the `ModemManager` attempts to use Teensy as a modem.

The [49-teensy.rules](https://www.pjrc.com/teensy/49-teensy.rules) include the following line which disables `ModemManager` when a Teensy is connected.

```
ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="04[789B]?", ENV{ID_MM_DEVICE_IGNORE}="1"
```

Would it be possible to include the Teensy udev rules in ResinOS? I think that the responsibility of `meta-resin` as I have seen reference to `60-resin-update-state.rules` in the core recipes. If this is not the correct place to open this issue then please let me know.

Perhaps it would be better to create RESIN_HOST_CONFIG_MM_DEVICE_IGNORE_VENDOR_ID and RESIN_HOST_CONFIG_MM_DEVICE_IGNORE_PRODUCT_ID as host configuration variables. Then in the case of a Teensy these could be set to `16c0` and `04[789B]?`

floion, 2018-04-30:
Hi,
correct, this should be done in meta-resin in case other boards may want to use a Teensy board.
I'm not sure if we want to make the supervisor handle these. @pcarranzav what is your take on making the supervisor be able to do that?
In the meanwhile @hpgmiskin you can create a PR in meta-resin to add such an udev rule in /etc/udev/rules.d/ to ignore such as it was done here: https://github.com/resin-os/resin-edison/pull/103

hpgmiskin, 2018-05-01:
@agherzan I think creating a `resin-extra-udev-rules`  recipe in order to contain the Teensy udev rules makes sense. Would this be best placed within the `recipes-connectivity` folder?

closed 2018-05-02 by agherzan

hpgmiskin, 2018-08-13:
I am getting the following error in `dmesg` output meaning there is a problem with the above fix.

```
[Fri Aug 10 19:30:07 2018] systemd-udevd[31]: starting version 215
[Fri Aug 10 19:30:07 2018] udevd[31]: invalid key/value pair in file /etc/udev/rules.d/49-teensy.rules on line 1,starting at character 1 ('!')
[Fri Aug 10 19:30:07 2018] udevd[31]: invalid key/value pair in file /etc/udev/rules.d/49-teensy.rules on line 2,starting at character 1 ('h')
[Fri Aug 10 19:30:07 2018] udevd[31]: invalid key/value pair in file /etc/udev/rules.d/49-teensy.rules on line 3,starting at character 1 ('t')
[Fri Aug 10 19:30:07 2018] udevd[31]: invalid key/value pair in file /etc/udev/rules.d/49-teensy.rules on line 4,starting at character 1 ('/')
[Fri Aug 10 19:30:07 2018] udevd[31]: invalid key/value pair in file /etc/udev/rules.d/49-teensy.rules on line 5,starting at character 1 ('h')
[Fri Aug 10 19:30:07 2018] udevd[31]: invalid key/value pair in file /etc/udev/rules.d/49-teensy.rules on line 6,starting at character 1 ('p')
[Fri Aug 10 19:30:07 2018] udevd[31]: invalid key/value pair in file /etc/udev/rules.d/49-teensy.rules on line 7,starting at character 1 ('h')
[Fri Aug 10 19:30:07 2018] udevd[31]: invalid key/value pair in file /etc/udev/rules.d/49-teensy.rules on line 8,starting at character 1 ('a')
[Fri Aug 10 19:30:07 2018] udevd[31]: invalid key/value pair in file /etc/udev/rules.d/49-teensy.rules on line 9,starting at character 1 ('/')
```

However when I attempt to `cat  /etc/udev/rules.d/49-teensy.rules` there is no such file or directory on the HostOS terminal in resin. Where might I locate the udev rules to check what is causing the error? I cannot see any matching characters in the intended udev rules so it could be that the file is being modified at some stage?

agherzan, 2018-08-14:
@hpgmiskin The file is actually installed in `/lib/udev/rules.d/49-teensy.rules`, see [link](https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-core/resin-extra-udev-rules/resin-extra-udev-rules.bb#L14). What version of the OS are you using and on what device? Are you using a custom build? If yes, what modifications have you done?

hpgmiskin, 2018-08-14:
We are running 2.13.6 on a raspberry pi. There is nothing custom about our OS build but running multi container stack. The Teensy rules are where you suggested and exactly the same as in meta resin.

agherzan, 2018-08-14:
@hpgmiskin Can you open a support ticket and share a device on which we can see this issue? I tried to reproduce and was not able to.