balena-os/meta-balena #1068: Integrate DNS with ntp server

agherzan, 2018-05-03:
Based on comments from arch call 2018-05-03

We are currently using systemd's timesyncd for as an NTP daemon. Find a way to integrate it with DHCP advertised NTP servers. If that is not possible, investigate how systemd's timesyncd compares to ntpd which has the needed integration available.

bbinet, 2018-05-04:
It was also considered to switch to Chrony as an NTP client in issue #860.

agherzan, 2018-05-04:
Good point, @bbinet . I realised I had a typo as well in the issue title. Clarified it. Are you aware of this functionality available in chrony?

agherzan, 2018-05-04:
On a second thought, I think both can be implemented using the NM dispatcher functionality.

ZubairLK, 2018-05-08:
In dhclient here->
https://github.com/resin-os/meta-resin/blob/301f10623edfec6dd17cb974f88f1a638d0d261d/meta-resin-common/recipes-connectivity/dhcp/files/dhclient.conf#L3

my laptop's Ubuntu dhclient.conf contains the following. Notice the last one is ntp-servers. dhclient can request this info from a router
```
        request subnet-mask, broadcast-address, time-offset, routers,
	domain-name, domain-name-servers, domain-search, host-name,
	dhcp6.name-servers, dhcp6.domain-search, dhcp6.fqdn, dhcp6.sntp-servers,
	netbios-name-servers, netbios-scope, interface-mtu,
	rfc3442-classless-static-routes, ntp-servers;
```

agherzan, 2018-05-08:
That is the way to get the info with dhcp client but we need to not only fetch this information but also make the NTP client honor it.

closed 2018-10-19 by agherzan