balena-os/meta-balena #1069: How to Install ResinOS in vmware?

rgaufman, 2018-05-03:
I tried to do this:

```
qemu-img convert resinos.img -O vmdk resinos.vmdk
```

1) In the VMware Fusion menu > New... > More Options > Create a custom virtual machine
2) Select Other Linux 4.x 64-bit as OS
3) Use an existing virtual disk > Select 'resinos.vmdk' as disk > Continue > Customize Settings > Save
4) In the VM setting panel, select the Hard Disk > Advanced options >Set 'Bus type': IDE > Apply settings changes
5) Start VM

The virtual machine starts and I can ssh using ```ssh root@resin.local -p22222``` but it takes me to a busy box that doesn't have docker or anything else.

I'm guessing this is because ResinOS is designed to boot and then install itself onto another hard drive. Is there a workaround to make this work in VMWare or alternatively an image version that is self-contained rather than an installer?

agherzan, 2018-05-04:
Are you using a NUC image? If yes, you are right. The default image is wrapped into a flasher which is intended to be burned on the internal device. You can use https://github.com/resin-os/resin-image-flasher-unwrap to get the image which you can boot directly.

closed 2018-05-14 by agherzan

rgaufman, 2018-05-15:
Sorry for the late reply, that worked great, thank you :)

nfaf1, 2019-01-07:
Hello :)
I have tried to do this as well to test it and when I boot the VM it gives me lots of errors:
[image]

Can you help please ?

nfaf1, 2019-01-07:
just a little more info ...

[image]

Please advise 

Thank you

agherzan, 2019-01-08:
Hi @nfaf1. It looks like the image you are booting doesn't have the required labels. How did you get and manipulate the image for VMware? Are you using a flasher or unpacked and ran a normal image directly?

nfaf1, 2019-01-08:
Hello Andrei, I grabbed the NUC image and ran the unwrapper as you instructed above 
The resulting image I converted to a vmdk which I then created a vm é actky like the first post 
The result it this 
I was made aware of a virtualbox environment which I manage to get up and running by using 2 disks where the second one is the flasher image 
Please advise 
Thank You 
Nuno Ferreira

nfaf1, 2019-01-08:
Yes, the setup with Virtualbox as per this document - https://www.balena.io/blog/no-hardware-use-virtualbox/ 
Works fine
The VMWare gives that problem. Please note that in VMWare I didn't do the 2 disk trick and instead I just unwrapped the flasher image (the one downloaded from your balena account) and converted it into VMDK disk and use it directly ... This gave me the output I sent you 

I really wanted to run this on Fusion but it appears that, unless @rgaufman did something not documented in this thread, the partitions screw up the boot in VMWare

Let me know your thoughts 

Thanks

nfaf1, 2019-01-08:
@agherzan another quick one .. the ssh command above doesn't work for me. How can I ssh into the box ?

richbayliss, 2019-01-08:
@nfaf1 I have just run through this myself as I wanted to have 2.29 running on my ESXi machine and I will detail the steps below that worked:

- downloaded the Intel Nuc 2.29 Development image
- used the `./docker-run ~/Downloads/balenaOS-nuc-2.19.0+rev1.dev.img` to unwrap it
- used the `./qemu-img convert output/resin-image.img -O vmdk output/balenaOS-2.29.0+rev1.dev.vmdk` to convert it to VMDK
- created a VM in ESXi using this image
  - boot mode was EFI
  - disk attached as Master on IDE:1
  - no other disk on the system
- booted and confirmed OK

nfaf1, 2019-01-08:
yes that was exactly what I did and I got the output described above
I am using a mac so I am using fusion

nfaf1, 2019-01-08:
I noticed that when I attach the disk from the VMDK file it comes as SCSI ... and if I load this in an ESXi server it gives me a scsi error

How did you overcome this ? How did you attach your vmdk as an ide disk ?

Thanks

nfaf1, 2019-01-08:
OK I found the issue. I was adding it in Fusion (mac) and it adds the hard drive as SCSI
This gives the SCSI issue when loading it in ESXi
I have added the VM in the vCentre and it added as an IDE and all was fine
With fusion for mac itself, the problem is the same

Thanks for your help guys

codewithcheese, 2020-01-28:
A note for future readers. I had to resize the disk Settings -> Hard Disk -> Expand Disk (in vmware workstation) to make the disk large enough to deploy the multi container sample application.