balena-os/meta-balena #1072: dns-search and dns suffix from router do not update /etc/resolv.conf

ZubairLK, 2018-05-07:
My wifi router appends a dns suffix .lan to hosts.
e.g. my laptop with a hostname zubair-xps-resin has a full hostname zubair-xps-resin.lan
Any device on the local network should be able to ping zubair-xps-resin.lan

A nifty trick ubuntu does (on my laptop) is that if the router broadcasts there is a dns suffix, ubuntu appends 'search $dns_suffix' to /etc/resolv.conf 

This allows the computer to automatically resolve zubair-xps-resin to zubair-xps-resin.lan.
Windows does something similar as well. so pinging zubair-xps-resin 'just works'

resinos doesn't seem to automatically append the dns suffix to /etc/resolv.conf
And the surprising part, I tried to manually configure it via network manager but failed too.

Here is what I tried

```
root@58170ef:~# cat /etc/resolv.conf 
# we use dnsmasq at 127.0.0.2 so that user containers can run their own dns cache and forwarder and not conflict with dnsmasq on the host
nameserver 127.0.0.2
root@58170ef:~# nmcli con show resin-wifi-01 | grep  ipv4.dns-search
ipv4.dns-search:                        lan
root@58170ef:~# cat /etc/NetworkManager/system-connections/resin-wifi-01  | grep -C2 dns
[ipv4]
method=auto
dns-search=lan
root@58170ef:~# ping zubair-xps-resin
ping: bad address 'zubair-xps-resin'
root@58170ef:~# ping zubair-xps-resin.lan
PING zubair-xps-resin.lan (192.168.1.249): 56 data bytes
64 bytes from 192.168.1.249: seq=0 ttl=64 time=3.415 ms
64 bytes from 192.168.1.249: seq=1 ttl=64 time=49.350 ms
^C
--- zubair-xps-resin.lan ping statistics ---
2 packets transmitted, 2 packets received, 0% packet loss
round-trip min/avg/max = 3.415/26.382/49.350 ms
root@58170ef:~# cat /etc/os-release 
ID="resin-os"
NAME="Resin OS"
VERSION="2.12.6+rev1"
VERSION_ID="2.12.6"
PRETTY_NAME="Resin OS 2.12.6+rev1"
RESIN_BOARD_REV="c1ac9a0"
META_RESIN_REV="b744564"
SLUG="raspberrypi3"
MACHINE="raspberrypi3"
VARIANT="Development"
VARIANT_ID="dev"
root@58170ef:~# 
```

This seems innocent. But there are customers who have resin devices on internal company networks (which usually add an internal suffix e.g. device.london.company.org, device.leeds.company.org)
finding this a bit painful. e.g. https://forums.resin.io/t/ping-devices-on-the-local-network-via-hostname/2082/23

And this might manifest in other funny ways. e.g. Does the app container inherit resolv.conf from the host? The missing dns search bug might then trickle into the app container.

ZubairLK, 2018-05-08:
@agherzan any thoughts?
I think dns-search is missing from here and perhaps a few other lines needed elsewhere.
https://github.com/resin-os/meta-resin/blob/301f10623edfec6dd17cb974f88f1a638d0d261d/meta-resin-common/recipes-connectivity/dhcp/files/dhclient.conf#L3

ZubairLK, 2018-05-08:
Looks like I missed dnsmasq with all the interesting symlinks and went down a rabbithole..

For future reference. The configuration at this point is
```
root@ab6c3f3:~# ls -al /etc/resolv*
-rw-r--r-- 1 root root  159 Feb 12 19:15 /etc/resolv-conf.dnsmasq
lrwxrwxrwx 1 root root   24 Mar 20 04:26 /etc/resolv.conf -> /etc/resolv-conf.dnsmasq
lrwxrwxrwx 1 root root   44 Mar 13 12:52 /etc/resolv.dnsmasq -> /var/run/resolvconf/interface/NetworkManager
root@ab6c3f3:~# ps  | grep dnsma
  591 nobody    2200 S    /usr/bin/dnsmasq -x /run/dnsmasq.pid -a 127.0.0.2,10.114.102.1 -7 /etc/dnsmasq.d/ -r /etc/resolv.dnsmasq -z --servers-file=/run/dn
root@ab6c3f3:~#
```

/etc/resolv.conf -> /etc/resolv-conf.dnsmasq 
which has the nameserver 127.0.0.2 entry.

/etc/resolv.dnsmasq -> /var/run/resolvconf/interface/NetworkManager
which has the network manager entries.
search lan
nameserver 192.168.1.1
nameserver 8.8.8.8

and dnsmasq runs with -r /etc/resolv.dnsmasq so any query is passed on.

closed 2018-05-08 by ZubairLK

agherzan, 2018-05-08:
Sorry @ZubairLK - late to the party.

raghav-cyb, 2020-01-22:
@ZubairLK  Even if /etc/resolv.dnsmasq has dns search domain entry, BalenaOS doesn't append to the for example ping 'hostname' resulting in ping: bad address

But works fine when domain name is explicitly specified like : ping hostname.domain

Any solution for this ?