balena-os/meta-balena #1081: plymouthd doesn't quit after displaying boot splash

ZubairLK, 2018-05-24:
plymouthd keeps running after displaying boot splash.

This prevents plymouth-reboot or plymouth-poweroff service from starting plymouthd and attaching to tty1 to display the splash on reboot and shutdown.

```
root@5ab792e:~# ps | grep ply
  590 root     42600 S    @sbin/plymouthd --tty=tty1 --mode=boot --pid-file=/run/plymouth/pid --attach-to-session --kernel-command-line=plymouth.ignore-serial-consoles splash
 1851 root      2788 S    grep ply
root@5ab792e:~#
```
only after killing this process (or `plymouth --quit`) does the splash screen show on a reboot/shutdown.

alexgg, 2020-02-19:
Related support thread. On occasions plymouth retains DRM master mode and prevents X from starting. Might be a regression. Details:
https://www.flowdock.com/app/rulemotion/public-s-premium/threads/wXyHYXONnVIeZ13e07JUQocY89U

alexgg, 2020-04-15:
We can assume that we are dealing with devices that have graphical output so they require a smooth transition between splash screen and graphical applications. To use X11 as an example, a smooth transition would look like:
```
plymouthd --mode=boot
plymouth show-splash
plymouth --wait (other services sync to this)
plymouth deactivate
(X starts)
(gdm starts)
plymouth quit --retain-splash 
....
plymouthd --mode=shutdown
plymouth show-splash
```
Now, because we don't control the application, the current behaviour which is just doing the --wait and leaving the rest to the application container seems correct to me.
We might want to provide an application example that correctly deals with plymouth, but that is outside of the scope of meta-balena so I am closing this.
Let's re-open if there is a customer request for any change.

closed 2020-04-15 by alexgg