balena-os/meta-balena #1092: resinOS filesystem label conflicts

agherzan, 2018-06-11:
resinOS uses filesystem labels based on which it mounts at boot time the partitions of the OS. Currently, we have:

- resin-boot
- resin-rootA
- resin-rootB
- resin-state
- resin-data

The mount services handling these mounts exclusively use these filesystem labels. The problem arises on systems where there are multiple block devices flashed with resinOS available at boot. Such of an example is a resinOS device which is meant to flash/multiply resinOS images by cloning an attached block device.

agherzan, 2019-04-16:
We will do this with a systemd service (but after boot partition mount so we can write down the generated UUIDs) with a couple of ConditionKernelCommandLine set. For example ConditionKernelCommandLine=!uuid_resin_boot. This service will translate from cmdline to filesystem UUID setting them accordingly. Also all the partition mounts will first try to use cmdline for mounting a partition and if the UUID is not available, fallback to labels. The last bit is the bootloader which will need to pass on the resinEnv UUIDs (when available) through cmdline.

So at first boot, uboot will not find anything expected in resinEnv so it won't pass anything related in cmdline. Initramfs mount will fallback to labels, systemd will enable the uuid generation service which will start because the conditions on cmdline are satisfied. During its execution will generate UUIDs, set them for filesystems and write them to boot partition (which was mounted falling back to labels).
On next boots, uboot will pass the resinEnv UUIDs to linux through cmdline. Initramfs mount will use them for mounting its partitions. Later on systemd will not start the uuid generation service as the conditions are not satisfied anymore and the partition mount services will use the cmdline uuids.

alexgg, 2020-02-17:
**Proposed test plan**

On a device using u-boot like raspberrypi4-64:

[ ] Initial boot does not pass uuids in command line and the initramfs generates them
[ ] Consequent boots read the uuids from file and passes them to boot. Initramfs does not generate them.
[ ] Booting an installed system with a plugged in device with conflicting labels mounts the correct uuids
[ ] The above works for both dos and gpt partition types

On a grub device like intel-nuc:

[ ] Initial boot does not pass uuids in command line and the initramfs generates them
[ ] Consequent boots read the uuids from file and passes them to boot. Initramfs does not generate them.
[ ] Booting an installed system with a plugged in device with conflicting labels mounts the correct uuids
[ ] The above works for both dos and gpt partition types

alexgg, 2020-02-27:
Related support thread https://app.frontapp.com/inboxes/teammates/957827/inbox/open/11023323331

commit ad1e4008 (2019-09-20): os-internals.md: Document file-system labels vs `UUIDs`

Fixes #1092

Change-type: minor
Chnagelog-entry: Add support for using file-system UUIDs in balenaOS
commit 2eab3c63 (2020-02-17): os-internals.md: Document file-system labels vs `UUIDs`

Fixes #1092

Change-type: minor
Chnagelog-entry: Add support for using file-system UUIDs in balenaOS
commit f2ccbd16 (2020-02-17): os-internals.md: Document file-system labels vs `UUIDs`

Fixes #1092

Change-type: minor
Chnagelog-entry: Add support for using file-system UUIDs in balenaOS
commit 48f63e9c (2020-02-19): os-internals.md: Document file-system labels vs `UUIDs`

Fixes #1092

Change-type: minor
Chnagelog-entry: Add support for using file-system UUIDs in balenaOS
closed 2020-04-17 by alexgg