balena-os/meta-balena #1099: Add config.json field to configure 'persistentLogging' log size

lekkas, 2018-06-14:
Currently there is an 8M limit on the journal log file (~3.5k entries). 

It would be very useful to have a way to override this setting, in cases where the default logrotate size is not enough to keep all required logs to troubleshoot an issue.

This could be done with an additional, optional `config.json` field.

Discussed in today's product call, not a high priority, but a nice to have if this is a low hanging fruit

Krismix1, 2018-08-08:
Good day
Do you have any updates on this?

agherzan, 2018-08-08:
Hi @krismix1, 

We don't have any progress on this and we might not be able to implement it very soon. We will post any updates here as soon as there is activity.

ZubairLK, 2019-01-22:
@agherzan 
the logs are in the state partition which makes creates an upper limit about how large the log can be. Perhaps we should move the logs to the data partition when doing this.

agherzan, 2019-01-22:
That is something to look into, I agree. Or maybe make the state larger.

ZubairLK, 2020-01-27:
Closing old/stale issues. Might still be valid. Can reopen if needed.

Check #1646

closed 2020-01-27 by ZubairLK