balena-os/meta-balena #1101: Add system-wide support for self-signed root CA certs

dfunckt, 2018-06-15:
In #1045 we added support for self-signed root certs in the Supervisor passing them over via config.json. A way is needed however for the given cert to be used by all services in resinOS. The proposed solution is to:

- [ ] read `.balenaRootCa` attribute from config.json and save the decoded string to a well-known location
- [ ] modify scripts/services that contact a Resin platform to use the cert when available
- [ ] modify `os-config` to use the passed cert for the initial contact with the platform and then install it to the well-known location if promotion succeeded

At any given time, there can be at most one cert installed.

Refs: https://www.flowdock.com/app/rulemotion/r-resinos/threads/CJiL2b8DCMsfWCwpncOyWnR3sr7

cc @agherzan @majorz

agherzan, 2018-11-02:
Discussed in 2018-11-02
We have two solutions here. For now, we will be configuring os-config crate which does the http request to use the custom certificate available in config.json (when available). On long term, the solution is to generate a rw layer with each os container and update/regenerate it with every hup. CC @majorz

majorz, 2018-12-28:
@agherzan is there are other services than os-config that need to use `balenaRootCA`, or we are good to close this issue?

agherzan, 2018-12-28:
I think we are good for now. Thanks.

closed 2018-12-28 by agherzan

imrehg, 2019-07-17:
@agherzan we are not good for now, `curl` needs this across the OS, for example for `resin-device-progress` as well

imrehg, 2019-07-17:
I believe it's continued in #1398

agherzan, 2019-07-17:
That is correct. The long term solution will be implemented as part of the issue you mentioned.