balena-os/meta-balena #1108: Second build fails

splitice, 2018-06-29:
The return value of groupadd and useradd is non zero in https://github.com/resin-os/meta-resin/blob/97a39066ec5b25d9cebb3382ef9b05214072f4d7/meta-resin-common/recipes-containers/docker-disk/files/entry.sh#L17 as this action has already been performed.

This is not an issue if executed in the docker container as each build is run from scratch, however if the build is executed outside of docker this will occur.

An easy fix is to add ``` || true``` to ignore the return value.

agherzan, 2018-07-12:
Hello @splitice . Sorry for coming to you a little late. I think your recommendation would be fine for this case. There is a big PR which touches quite of a bunch of the docker-disk recipes and I will try to include this as well https://github.com/resin-os/meta-resin/pull/1028 . 

Wouldn't you need a similar fix for the `useradd` command as well?

splitice, 2018-07-13:
Yes @agherzan 

Heres the fix we employed: https://github.com/HalleyAssist/meta-resin/commit/5305d8eb7628cd8744262ed911ff372b6d2294ca

agherzan, 2018-07-13:
@splitice I backported that commit in the above PR. Tried to keep the author information in the commit log but tell me if I missed anything. When that PR gets merged, this issue will get closed automatically.

splitice, 2018-07-13:
I think you have it covered. Have a good weekend mate.

agherzan, 2018-07-13:
You too @splitice!

closed 2018-07-17 by agherzan