balena-os/meta-balena #1132: mount.aufs seg faults in sumo

ZubairLK, 2018-07-25:
Running mount.aufs seg faults in sumo 

```
root@8b46e19:/tmp# cat /etc/os-release
ID="resin-os"
NAME="Resin OS"
VERSION="2.13.6+rev1"
VERSION_ID="2.14.0"
PRETTY_NAME="Resin OS 2.13.6+rev1"
RESIN_BOARD_REV="a9400f1"
META_RESIN_REV="fe3eb02"
SLUG="raspberrypi3"
MACHINE="raspberrypi3"
VARIANT="Development"
VARIANT_ID="dev"
root@8b46e19:/tmp# mount.aufs
Segmentation fault (core dumped)
root@8b46e19:/tmp#
```

And a working one for BBB 2.12.5

```
root@5a524b5:~# cat /etc/os-release
ID="resin-os"
NAME="Resin OS"
VERSION="2.12.5+rev1"
VERSION_ID="2.12.7"
PRETTY_NAME="Resin OS 2.12.5+rev1"
RESIN_BOARD_REV="63c3e77"
META_RESIN_REV="b9b070f"
SLUG="beaglebone-black"
MACHINE="beaglebone"
VARIANT="Development"
VARIANT_ID="dev"
root@5a524b5:~# mount.aufs
aufs-util for aufs3.14 and later 20150323
mount.aufs:mount.aufs.c:165: : Invalid argument
root@5a524b5:~#
```

ZubairLK, 2018-07-25:
Another datapoint pi v2.12.5
```
root@519679d:~# cat /etc/os-release
ID="resin-os"
NAME="Resin OS"
VERSION="2.12.5+rev2"
VERSION_ID="2.12.5"
PRETTY_NAME="Resin OS 2.12.5+rev2"
RESIN_BOARD_REV="c0333a9"
META_RESIN_REV="5c6ee83"
SLUG="raspberrypi3"
MACHINE="raspberrypi3"
VARIANT="Development"
VARIANT_ID="dev"
root@519679d:~# mount.aufs
aufs-util for aufs4.4 20170327
mount.aufs:mount.aufs.c:268: : Invalid argument
root@519679d:~#
```

agherzan, 2018-07-31:
@ZubairLK can you please retry this with the current HEAD of meta-resin?

agherzan, 2018-10-19:
Fixed in https://github.com/resin-os/meta-resin/commit/969d4cb2865cc6f959450daaa60eb02e90ac4766#diff-d5c42a9d26ccc16b4a31eb2571f24a0e

closed 2018-10-19 by agherzan