balena-os/meta-balena #1140: Avahi interface description missing from /usr/share/dbus-1/interfaces/

lekkas, 2018-08-07:
Reported at https://forums.resin.io/t/2592

> It looks like the Avahi interface description document is not present under /usr/share/dbus-1/interfaces/. The dbus-native library relies on DBus' introspection description documents to allow access to the interfaces.

Device type: IntelNUC image on a Zotac computer

majorz, 2018-11-12:
Steps to reproduce:
```
dbus-send --system --type=method_call --print-reply --dest=org.freedesktop.Avahi / org.freedesktop.DBus.Introspectable.Introspect
```

Leads to:
```
Nov 12 17:08:14 518a272 avahi-daemon[697]: ../../avahi-0.7/avahi-daemon/chroot.c: open() failed: No such file or directory
Nov 12 17:08:14 518a272 avahi-daemon[697]: Failed to open /usr/share/dbus-1/interfaces/org.freedesktop.Avahi.Server.xml: Invalid argument
Nov 12 17:08:14 518a272 avahi-daemon[697]: Failed to load introspection data.
```

agherzan, 2018-11-13:
Thanks @lekkas and @majorz . I'll take a look at this and come back.

PR #1283 (2018-11-14): avahi: Bring back avahi dbus introspection files
Poky removes the dbus introspection description documents for avahi.
See:
http://git.yoctoproject.org/cgit.cgi/poky/commit/?id=59a08907eafffde664079b9a2068f47131dd3f5d

dbus-native node module requires this data to be available to allow
access to interfaces.

Fixes #1140

Change-type: minor
Changelog-entry: Include avahi d-bus introspection files in rootfs

---
### Contributor checklist

- [x] Changes have been tested
- [x] `Change-type` present on at least one commit
- [x] `Signed-off-by` is present
- [x] The PR complies with the [Open Embedded [... cut]
commit 4703b92a (2018-11-14): avahi: Bring back avahi dbus introspection files

Poky removes the dbus introspection description documents for avahi.
See:
http://git.yoctoproject.org/cgit.cgi/poky/commit/?id=59a08907eafffde664079b9a2068f47131dd3f5d

dbus-native node module requires this data to be available to allow
access to interfaces.

Fixes #1140

Change-type: minor
Changelog-entry: Include avahi d-bus introspection files in rootfs
commit 551734e4 (2018-11-16): avahi: Bring back avahi dbus introspection files

Poky removes the dbus introspection description documents for avahi.
See:
http://git.yoctoproject.org/cgit.cgi/poky/commit/?id=59a08907eafffde664079b9a2068f47131dd3f5d

dbus-native node module requires this data to be available to allow
access to interfaces.

Fixes #1140

Change-type: minor
Changelog-entry: Include avahi d-bus introspection files in rootfs
commit d98f0d47 (2018-11-23): avahi: Bring back avahi dbus introspection files

Poky removes the dbus introspection description documents for avahi.
See:
http://git.yoctoproject.org/cgit.cgi/poky/commit/?id=59a08907eafffde664079b9a2068f47131dd3f5d

dbus-native node module requires this data to be available to allow
access to interfaces.

Fixes #1140

Change-type: minor
Changelog-entry: Include avahi d-bus introspection files in rootfs
closed 2018-11-19 by agherzan